gazebosim/gz-sim #70: GUI Commands Tutorial

JShep1, 2020-04-20:
Create tutorials documenting the GUI tools currently available:

* Key bindings (similar to Gazebo's hotkeys http://gazebosim.org/hotkeys.html)
* Transform control
* View angle widget
* Align tool
* Model insertion from Fuel

I imagine we can do a first pass just to make sure they're documented, and improve on them later. It's better to have some documentation than none, so we don't need to go too crazy at first.

maryaB-osr, 2020-04-20:
https://github.com/ignitionrobotics/docs/pull/10

I opened a PR for this over in /docs. I guess we can leave this issue here and just link them?

chapulina, 2020-05-15:
They're live:

https://ignitionrobotics.org/docs/blueprint/gui

https://ignitionrobotics.org/docs/citadel/gui

closed 2020-05-15 by chapulina