gazebosim/gz-sim #84: Add helper API to encapsulate the check/create step for components

gerkey, 2020-04-21:
In the course of migrating a plugin from Gazebo to Ignition (#80), I ended up following a pattern like this when accessing components:
```
if(!_ecm.EntityHasComponentType(this->dataPtr->modelLink, components::WorldPose::typeId))
{
  _ecm.CreateComponent(this->dataPtr->modelLink, ignition::gazebo::components::WorldPose());
}
```
That is, do some sort of lookup and if it fails, then create the component, and subsequently read from or write to it.

This pattern seems common enough that it would be worthwhile providing helper API to encapsulate it into a single `PleaseGiveMeThisComponentButCreateItFirstIfYouNeedTo()` call.

On a related topic, and also related to defining each system's component API (#83), it would be useful to know which components, if any, are guaranteed to exist. I found that sometimes I needed to do the check-create pattern and sometimes I could just use a component blindly without checking its validity. If that's just bad practice, and the right thing to do is to always check for existence of a component before accessing it, then we should say so in the docs (and also clean up internal component handling, e.g., the [IMU system accesses the `Name` component without checking that it's valid](https://github.com/ignitionrobotics/ign-gazebo/blob/master/src/systems/imu/Imu.cc#L161-L162))

peci1, 2020-10-02:
The example Brian showed gets even more obscure when you need to set some value to the component:

https://github.com/ignitionrobotics/ign-gazebo/blob/9fb057712d5eb1c4965b2755bfb17e7ce6234fe3/src/systems/joint_position_controller/JointPositionController.cc#L231-L241

From the developer's point of view, this is nightmare...

Compare with Gazebo:

```c++
this->dataPtr->joints[iter->first]->SetForce(0, iter->second);
```

I think this design pattern (or rather antipattern?) really needs some simplification. The earlier it gets implemented, the less work will be spent on rewriting other parts of the codebase :) Let's make it a challenge for Edifice? :)

One good solution would be to provide wrappers like https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo4/src/Link.cc for most of the relevant entity types. I always wondered why there is Link.cc and not Joint.cc ...

chapulina, 2020-10-02:
> One good solution would be to provide wrappers 

That's in the works, see #375 #378 #325

> Let's make it a challenge for Edifice?

The goal is to add these from Citadel onwards :wink:

peci1, 2020-10-02:
Great to hear that!

chapulina, 2020-10-31:
See https://github.com/ignitionrobotics/ign-gazebo/pull/436 for a helper function to create and set component values.

peci1, 2020-10-31:
Hooray!

gerkey, 2020-11-03:
Looking good! Can we also have a `GetComponentData()` helper?

chapulina, 2020-11-04:
> Can we also have a GetComponentData() helper?

That was added in https://github.com/ignitionrobotics/ign-gazebo/pull/378 :wink:

gerkey, 2020-11-04:
Got it! Unless something else is meant to be done, then I think that this issue can be closed.

closed 2020-11-17 by chapulina