gazebosim/gz-sim #85: Simplify determination of a sensor's topic name

gerkey, 2020-04-21:
In the course of migrating a plugin from Gazebo to Ignition (#80), I needed to subscribe to data published via `ign-transport` by an IMU sensor. To determine the fully qualified topic name to subscribe to, I ended up jumping through [quite some hoops](https://github.com/gerkey/ardupilot_gazebo/blob/ignition/src/ArduPilotPlugin.cc#L841-L884) inside `PreUpdate()`:

```
// This lookup is done in PreUpdate() because in Configure() it's not possible to get the fully qualified topic name we want
  if(!this->dataPtr->imuInitialized)
  {
    // Set unconditionally because we're only going to try this once.
    this->dataPtr->imuInitialized = true;
    std::string imuTopicName;
    _ecm.Each<ignition::gazebo::components::Imu, ignition::gazebo::components::Name>(
            [&](const ignition::gazebo::Entity &_imu_entity,
                const ignition::gazebo::components::Imu * /*_imu*/,
                const ignition::gazebo::components::Name *_name)->bool
        {
          if(_name->Data() == this->dataPtr->imuName)
          {
            // The parent of the imu is imu_link
            ignition::gazebo::Entity parent = _ecm.ParentEntity(_imu_entity);
	    this->dataPtr->modelLink = parent;
            if(parent != ignition::gazebo::kNullEntity)
            {
              // The grandparent of the imu is the quad itself, which is where this plugin is attached
              ignition::gazebo::Entity gparent = _ecm.ParentEntity(parent);
              if(gparent != ignition::gazebo::kNullEntity)
              {
                ignition::gazebo::Model gparent_model(gparent);
                if(gparent_model.Name(_ecm) == this->dataPtr->modelName)
                {
                  imuTopicName = ignition::gazebo::scopedName(_imu_entity, _ecm) + "/imu";
                  igndbg << "Computed IMU topic to be: " << imuTopicName << std::endl;
                }
              }
            }
          }
          return true;
        });
    
    if(imuTopicName.empty())
    {
      ignerr << "[" << this->dataPtr->modelName << "] "
            << "imu_sensor [" << this->dataPtr->imuName
            << "] not found, abort ArduPilot plugin." << "\n";
      return;
    }

    this->dataPtr->node.Subscribe(imuTopicName, &ignition::gazebo::systems::ArduPilotPluginPrivate::imuCb, this->dataPtr.get());
  }
```

I don't know what the right answer looks like, but that seems to me too much work to get to a topic name.

chapulina, 2020-12-31:
I'm closing this because the topic can be found out from the `SensorTopic` component since #266.

The example above could become:

```
    auto imuEntity = _ecm.EntityByComponents(
        components::Name(this->dataPtr->imuName),
        components::Imu(),
        components::ParentEntity(this->dataPtr->modelLink));
    auto imuTopicName = _ecm.ComponentData<components::SensorTopic>(imuEntity);
```

azeey, 2022-11-05:
Reopening since `SensorTopic` is not available for rendering sensors.

closed 2023-02-27 by iche033