gazebosim/gz-sim #89: Load worlds from Fuel

chapulina, 2020-04-23:
Ignition Fuel hosts world files as well as models:

https://app.ignitionrobotics.org/fuel/worlds

Ignition Gazebo doesn't have a nice way of loading them yet though. It would be nice to support loading them from the command line like:

`ign gazebo https://fuel.ignitionrobotics.org/1.0/openrobotics/worlds/demo.sdf`

We could also consider moving the [example worlds](https://github.com/ignitionrobotics/ign-gazebo/tree/master/examples/worlds) to Fuel.

chapulina, 2020-08-05:
Just summarizing some of the conversations that happened around this issue:

* Worlds on Fuel are directories, not single files
* Unlike models, which have `model.config`, we don't have a standard way of setting metadata for worlds
* For now, users should pass the world root directory URL and we'll load the first `.sdf` file found
* We need to update the [worlds we currently have on Fuel](https://app.ignitionrobotics.org/fuel/worlds) to use `.sdf` instead of `.world`. As a first step, we could have both files there to avoid breaking CI.
* Going forward, we're considering supporting metadata for worlds. @nkoenig will work on a short design document for that.

JShep1, 2020-08-06:
Both PRs resolving this issue can be found at https://github.com/ignitionrobotics/ign-gazebo/pull/274 and https://github.com/ignitionrobotics/ign-launch/pull/43

JShep1, 2020-08-24:
Closing this issue as the PR fixing the issue (https://github.com/ignitionrobotics/ign-gazebo/pull/274) has been merged

PR #274 (2020-08-14): Add Fuel World Support
This PR adds functionality for the user to specify a fuel world URL after `ign gazebo`, ie, `ign gazebo https://staging-fuel.ignitionrobotics.org/1.0/gmas/worlds/ShapesClone`

It will first check for a cached copy of the world, and then download the world if no cached copy is found.  After this, it will search for an sdf file on the locally downloaded/cached path and load the first one found.

See also https://github.com/ignitionrobotics/ign-launch/pull/43

Closes #89
closed 2020-08-24 by JShep1