gazebosim/gz-sim #110: Visualize center of mass

chapulina, 2020-04-29:
Like on Gazebo-classic, it should be possible to visualize links' CoMs so users can debug their models and simulations.

On classic, CoM visuals consist of a sphere scaled proportionally to the link's mass, and lines stretching to the link's bounding box. The sphere represents a uniform ball of lead with the same mass, so if it is as big as your model, your model is as dense as lead!

CoM visualization could be toggled from:

* A world's right-click context menu: toggles all CoMs in the world
* A model's right-click context menu: toggles all CoMs in the model
* A link's right-click context menu

We could also consider adding toggle buttons to an entity's component inspector.

[Here](https://github.com/osrf/gazebo/blob/gazebo11/gazebo/rendering/COMVisual.cc)'s classic's `CoMVisual` implementation as a reference.

And [here](https://osrf-migration.github.io/gazebo-gh-pages/#!/osrf/gazebo/pull-requests/1445/page/1)'s the PR where the visuals were added to Gazebo-classic.

[image]

ahcorde, 2021-07-05:
Ign-rendering6 adds this feature https://github.com/ignitionrobotics/ign-rendering/pull/345

next step create a PR on ign-gazebo

FYI @atharva-18

chapulina, 2021-07-06:
#903

PR #903 (2021-07-06): Visualize center of mass
# 🎉 New feature

Closes #110 

## Summary
Adds the ability to visualize the center of mass, similar to Gazebo Classic. Related to #861.
[image]

## Test it
Test it by right-clicking on an entity and selecting `View -> Center of Mass`.

## Checklist
- [x] Signed all commits for DCO
- [ ] Added tests
- [ ] Added example and/or tutorial
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [x] `codecheck` passed (See [contributing](https://ignitionrobotics.org/docs/all/contributing#contributing-code))
- [ ] All tests passed (See [test coverag [... cut]
closed 2021-07-06 by chapulina