gazebosim/gz-sim #112: Visualize contacts

chapulina, 2020-04-29:
Gazebo-classic has an option to enable visualization of all contacts in the world as blue spheres with a green line indicating the force's direction and magnitude. These help users debug their model's surface properties and and identify contact points while stepping through simulation.

Ignition Gazebo already has a [Contact system](https://github.com/ignitionrobotics/ign-gazebo/tree/master/src/systems/contact) which populates the [ContactSensorData](https://github.com/ignitionrobotics/ign-gazebo/blob/master/include/ignition/gazebo/components/ContactSensorData.hh) component.

[Here](https://github.com/osrf/gazebo/blob/6fd426b3949c4ca73fa126cde68f5cc4a59522eb/gazebo/rendering/ContactVisual.cc)'s Gazebo-classic's `ContactVisual` implementation.

[image]

mcres, 2020-06-26:
I'll take this, as this is partially addressed in #229.

My idea is writing a plugin with a checkbox that allows the user to turn on and off the visualization of the contacts. This plugin would be linked to a source file that is continually checking for new [ContactSensorData](https://github.com/ignitionrobotics/ign-gazebo/blob/master/include/ignition/gazebo/components/ContactSensorData.hh) components when the checkbox is on, and using the /marker service to actually visualize the contacts.
However, currently the contact messages only return the position. Work is going on in https://github.com/ignitionrobotics/ign-physics/pull/40.

Is this a good idea?

mabelzhang, 2020-06-29:
Hey @azeey, since you're the most familiar with the Contact plugin, is there a preference how we want to hook up the contact visualization to it (see suggestions 1-3 above)? Thanks!

azeey, 2020-06-29:
Option 1 seems reasonable to me. The GUI plugin should be able to get the contact sensor data from the GUI ECM (almost a copy of the server ECM). We might have to use `_contacts->SetData` and `_ecm.SetChanged` in https://github.com/ignitionrobotics/ign-gazebo/blob/9ab1cc3f17173c57028f3c460121a36cbdb0c8d7/src/systems/physics/Physics.cc#L1722

mcres, 2020-07-06:
@azeey Would it be currently possible to get the info about all of the contacts in the simulation without requiring the link to have a Contact sensor (i.e. looking for `ContactSensorData`)?. Given [this TODO comment](https://ignitionrobotics.org/api/gazebo/3.0/namespaceignition_1_1gazebo_1_1components.html#a1c664a3515fe885c6db0c32c826030a4), I understand that it is not.

I think that @chapulina meant getting _all_ the contacts, but I don't know if that's possible without modifying too much the physics plugin.

mcres, 2020-07-07:
Following the conversation with @azeey : the physics plugin fills the `components::ContactSensorData` for entities with both `ContactSensorData` and `Collision` components.

This means that getting information of `<collision>` contacts that don't have a `<sensor name="contact_sensor" type="contact">` is easily achievable by creating a `components::ContactSensorData` in the corresponding entity, which in turn will be filled by physics.

chapulina, 2021-05-05:
Closed by #234

closed 2021-05-05 by chapulina