gazebosim/gz-sim #113: Unload model plugins when the model is removed from the world

diegoferigo, 2020-04-29:
As discussed in https://github.com/ignitionrobotics/ign-gazebo/issues/39#issuecomment-613957221, model plugins are not unloaded when the model is removed from the world.

There are few problems that this behaviour could cause:

1. Memory leak. If the simulation is very long and models are inserted and deleted multiple times, the RAM consumption slowly increases. (e.g. https://github.com/ignitionrobotics/ign-gazebo/issues/47)
1. If the plugin access ECM components of the model and it does not handle properly the case when they are not found (because the model has been removed), the behaviour could be undefined.

In #39 the problem was acknowledged but no specific issue was opened to track it.

ejalaa12, 2022-12-22:
Hi, 
I can confirm that this problem is still present in gazebo Garden.
It's easily reproducable with an Hydrodynamics system which prints ["No linear vel"](https://github.com/gazebosim/gz-sim/blob/gz-sim7/src/systems/hydrodynamics/Hydrodynamics.cc#L446-L450) once its model has been removed from the world. Which means the system is still running, but the entity is not present anymore.

takahashi-e6, 2024-06-29:
Hi.

This problem seems to occur in Fortress as well, and can be reproduced by simply deleting the plugin-enabled model with the following command:

ign service -s /world/empty/remove --reqtype ignition.msgs.Entity --reptype ignition.msgs.Boolean --timeout 5000 --req 'type: MODEL; name: "my_model"'

How can I properly delete a model?

I looked up the ign command but couldn't find the answer.

MSZ98, 2025-01-17:
Hi,

In the `gazebo_ros_pkgs` -> `gazebo_ros_openni_kinect.cpp` -> `Load()` method, the `GazeboRosCameraUtils` plugin is loaded and stays active even after the model is removed from the simulation. When you try to add a model with a camera to the simulation again using the same namespace, `gazebo_ros_openni_kinect.cpp` attempts to load the `GazeboRosCameraUtils` plugin once more. However, an error occurs because the first `GazeboRosCameraUtils` plugin already created a dynamic reconfigure ROS service, and there’s an attempt to advertise the same service again.

I could use a different namespace for the new model, but this solution feels messy, and the old camera thread could still impact performance.