gazebosim/gz-sim #123: Insert models from local paths

iche033, 2020-05-01:
Currently models are downloaded from fuel (app.ignitionrobotics.org) and saved to fuel cache in `~/.ignition/fuel`.  The idea is to support pulling models locally without going through fuel.

Related design doc:
https://github.com/ignitionrobotics/design/pull/1

Related PRs:
https://osrf-migration.github.io/ignition-gh-pages/#!/ignitionrobotics/ign-fuel-tools/pull-requests/123/page/1
https://osrf-migration.github.io/ignition-gh-pages/#!/ignitionrobotics/ign-gazebo/pull-requests/532/page/1

nkoenig, 2020-05-26:
@JShep1 , we need this for SubT. Are you able to pick this up and target and MVP to blueprint?

JShep1, 2020-05-26:
@nkoenig Sure, I'll prioritize this issue and try to get started later today/tomorrow.

nkoenig, 2020-05-26:
Awesome, thanks!

chapulina, 2020-05-26:
Just clarified with @nkoenig and @JShep1 that I'll take this one and John will work on #31.

peci1, 2020-08-18:
Hasn't this been resolved? I can launch the SubT simulator pulling local models when I pass `localModel:=true`.

iche033, 2020-08-18:
I think for subt, if you don't have the model in fuel cache, the materials will still fail to load with `localModel:=true`. Related issue #286

chapulina, 2020-09-24:
I believe this has been solved and is available on:

* Blueprint since 2.24.0
* Citadel since 3.3.0
* Dome since 4.0.0~pre1

closed 2020-09-24 by chapulina