gazebosim/gz-sim #124: Install instructions using src for all ign libraries

catskul, 2020-05-04:
Instructions assume install of ign libraries from apt. I think if someone is installing from source they're likely to want to be installing the ign dependencies from source as well since trying to compile the latest might not be packaged.

Any change for these sort of build instructions? Or perhaps a meta repo with instructions for building all?

chapulina, 2020-05-04:
Hi @catskul , have you seen these? https://ignitionrobotics.org/docs/citadel/install_ubuntu_src

Or are you talking about building the upstream dependencies from source, like Qt, protobuf and so on?

catskul, 2020-05-04:
Have not. Thank you!

Would it make sense for me to put in a PR adding that link in the README.md?

chapulina, 2020-05-06:
Yeah that sounds good to me, thanks! Maybe a link at the beginning of the `Source Install` section.

CC @maryaB-osr

chapulina, 2021-11-15:
Done in https://github.com/ignitionrobotics/ign-gazebo/pull/1078

closed 2021-11-15 by chapulina