gazebosim/gz-sim #131: ign-gazebo regressions test failing INTEGRATION_rgbd_camera.test_ran

tfoote, 2020-05-11:
On linux: https://build.osrfoundation.org/job/ignition_gazebo-ci-master-bionic-amd64/
And MacOS: https://build.osrfoundation.org/job/ignition_gazebo-ci-master-homebrew-amd64/

Test `INTEGRATION_rgbd_camera.test_ran` is failing. 

`Expected: (depthBuffer) != (nullptr), actual: NULL vs 8-byte object <00-00 00-00 00-00 00-00>`

From: https://build.osrfoundation.org/job/ignition_gazebo-ci-master-bionic-amd64/lastCompletedBuild/consoleText

<details>

```
109: /var/lib/jenkins/workspace/ignition_gazebo-ci-master-bionic-amd64/ign-gazebo/test/integration/rgbd_camera.cc:95: Failure
109: Expected: (depthBuffer) != (nullptr), actual: NULL vs 8-byte object <00-00 00-00 00-00 00-00>
109/116 Test #109: INTEGRATION_rgbd_camera ..............................***Exception: SegFault  1.90 sec
```
</details>

chapulina, 2020-05-14:
I think this was fixed for `ign-gazebo2` on [this PR](https://osrf-migration.github.io/ignition-gh-pages/#!/ignitionrobotics/ign-gazebo/pull-requests/570/page/1), which is being forward-ported to `ign-gazebo3` on #136

j-rivero, 2020-05-20:
The job pointed by Tully checks the `master` branch of `ign-gazebo` and the PR landed in `ignition-gazebo3` branch which does not seems to suffer from the problem before/after merging the PR https://build.osrfoundation.org/job/ignition_gazebo-ci-ign-gazebo3-bionic-amd64/ . Should we prepare a forward to master branch?

chapulina, 2020-05-20:
> Should we prepare a forward to master branch?

This is happening on #144 :wink:

j-rivero, 2020-05-20:
> This is happening on #144 wink

confirmed, I don't see the problem in the CI of that PR.

PR #144 (2020-05-23): 3 ➡️ 4
@mabelzhang , I'm having a hard time getting the logging tests to pass. I'm opening this PR for visibility but I'm still looking into these tests. Let me know if you already have all the answers :wink: 

It's a bit confusing because we deprecated the `<path>`, but that's still the way we inject the path into the plugin, right? I think we could change this approach to simplify things a bit. For example, we could put the log record configuration into a component. We're doing something similar to propagate the physics engine from the command line to the physics plugin. If we take this rout [... cut]
closed 2020-05-23 by chapulina