gazebosim/gz-sim #141: master-homebrew-amd64 INTEGRATION_touch_plugin regression test segfault

tfoote, 2020-05-14:
A new test failure appeared tonight in the nightly dome builds

From: https://build.osrfoundation.org/job/ignition_gazebo-ci-master-homebrew-amd64/6/consoleFull

```
97: [ RUN      ] TouchPluginTest.MultiLink
97: [Msg] Loading SDF world file[/Users/jenkins/workspace/ignition_gazebo-ci-master-homebrew-amd64/ign-gazebo/test/worlds/touch_plugin.sdf].gazebo/test/worlds/touch_plugin.sdf].
97: Error [Param.cc:423] Unknown parameter type[]
97: Exception [Param.cc:70] SDF ASSERTION                     
97: Invalid parameter
97: In function       : Param
97: Assert expression : this->ValueFromString(_default)
 97/116 Test  #97: INTEGRATION_touch_plugin .............................***Exception: SegFault  6.65 sec
test 98
```

jacobperron, 2020-09-22:
Possibly related failure: https://build.osrfoundation.org/job/ignition_gazebo-ci-master-homebrew-amd64/32/testReport/

```
115/130 Test #115: INTEGRATION_touch_plugin .............................Child aborted***Exception:  10.11 sec
```

jacobperron, 2021-07-19:
Most recent failure: https://build.osrfoundation.org/job/ignition_gazebo-ci-ign-gazebo4-homebrew-amd64/81