gazebosim/gz-sim #154: Library error for [/usr/local/lib/libignition-gazebo3-ign.so.3.2.0]: can't load /usr/local/lib/libignition-gazebo3-ign.so.3.2.0

agayev169, 2020-05-25:
I get the following error when I run Ignition Gazebo built from source: Library error for [/usr/local/lib/libignition-gazebo3-ign.so.3.2.0]: can't load /usr/local/lib/libignition-gazebo3-ign.so.3.2.0. I have noticed that this error pops up with each library that I build. Actually, the only library that I need for now is ign-rendering, however, if I install all the ignition packages from binary, and install ignition rendering from source, I get the same error but with libignition-rendering. I have tried to build everything on 2 different PC's and got the same error both times. Is there anything I can do to solve this problem?

chapulina, 2020-05-25:
I suspect you are trying to use a bad combination of libraries.

> Ignition Gazebo built from source

Did you also build other dependencies from source? Which branch(es) are you building? Do you have any debs installed?

agayev169, 2020-05-26:
It seems that the problem was with ```$LD_LIBRARY_PATH```. To solve the problem you should add ```/usr/local/lib``` to your ```$LD_LIBRARY_PATH```.
So, 
```export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/lib```
must solve the problem.

I think that I can close the issue now. I still have a problem with ign-rendering, but it would be better to open an issue in the corresponding repo.

closed 2020-05-26 by agayev169