gazebosim/gz-sim #168: Fix compilation on Windows

chapulina, 2020-06-01:
Currently our Windows CI is disabled for `ign-gazebo` (and `ign-launch` on top of that) because it doesn't compile.

It may be an infrastructure issue, because the [error on Jenkins](https://build.osrfoundation.org/job/ign_gazebo-ign-2-win/78/consoleFull) says that dependencies failed to build:

```
Check that the following packages have been built:
- sdformat8
- ignition-physics1
- ignition-rendering2
- ignition-gui2
- ignition-sensors2
Failed   <<< ignition-gazebo2	[ Exited with code 1 ]
```

Although all these dependencies pass on their Windows CIs.

scpeters, 2020-10-09:
see also https://github.com/ignitionrobotics/ign-physics/issues/87

jacobperron, 2021-07-19:
I'm not sure if this ticket captures addressing test failures for Ediface and Fortress, but here's the latest status:

**Ediface:** https://build.osrfoundation.org/job/ign_gazebo-ign-5-win/24
- 204 test failures
- 128 build warnings

**Fortress:** https://build.osrfoundation.org/job/ign_gazebo-ci-win/109/
- 203 test failures
- 160 build warnings

traversaro, 2022-10-20:
A bit a recap of the situation, after https://github.com/gazebosim/gz-sim/pull/1764 :

## GUI support is disabled
`ign gazebo`  and `ign gazebo -g`  explicitly do not work, printing instead a nice message pointing to this issue : )

## server does not work out of the box if installed by conda-forge

`ign gazebo -s` seems to be working fine, but no extensive test have been performed. Furthermore, when ign gazebo is installed via conda-forge, if the machine is clean, `ign gazebo -s` will fail with an error similar to (see https://github.com/conda-forge/libignition-gazebo-feedstock/issues/32):
~~~
[Err] [D:\bld\libignition-gazebo6_1635251434674\work\src\ServerConfig.cc:900] Failed to copy installed config [D:\bld\libignition-gazebo6_1635251434674_h_env\Library\share\ignition\ignition-gazebo6\server.config] to default config [C:\Users\runneradmin.ignition\gazebo\6\server.config].(file D:\bld\libignition-gazebo6_1635251434674_h_env\Library\share\ignition\ignition-gazebo6\server.config doesn't exist)
~~~

**Pay attention that this problem is not reproducible if in your system `%HOMEDRIVE%%HOMEPATH%\.ignition\gazebo\6\server.config` already exists, for example because you successfully run `ign gazebo` once from a Ignition Gazebo installed from source on your machine.**

This happens due to the combination of two reasons:
* Ignition Gazebo/Gazebo Sim is not relocatable (see https://github.com/gazebosim/gz-sim/issues/626)
* The `IGN_GAZEBO_SERVER_CONFIG_PATH`/`GZ_SIM_SERVER_CONFIG_PATH` preprocessor variables that Ignition Gazebo/Gazebo Sim hardcodes inside its library, cannot be overwritten via environment variables (see https://github.com/conda-forge/libignition-gazebo-feedstock/pull/52#issuecomment-1285233271). Pay attention that there is an environment variable with that name, but it has a different meaning. So, it is not possible to tell to Gazebo Sim where to find at runtime the default `server.config` file, if the installation prefix is different from the one. This is a problem in general (see indeed https://github.com/gazebosim/gz-sim/issues/626, that is not Windows specific), but on conda Linux/macOS this problem is worked around as there is relocation also of strings contained in the library binary (see https://docs.conda.io/projects/conda-build/en/latest/resources/make-relocatable.html), but this functionality is not available on Windows.

An easy (but not robust) workaround is to manually copy the file, i.e. : 
~~~
cp %CONDA_PREFIX%\Library\share\ignition\ignition-gazebo6\server.config %HOMEDRIVE%%HOMEPATH%\.ignition\gazebo\6
~~~

traversaro, 2022-11-15:
> So, it is not possible to tell to Gazebo Sim where to find at runtime the default `server.config` file, if the installation prefix is different from the one. This is a problem in general (see indeed [...]

Related comment: https://github.com/gazebosim/gz-sim/issues/626#issuecomment-1315972993 .

traversaro, 2023-06-21:
In https://github.com/conda-forge/gz-physics-feedstock/pull/11 and https://github.com/conda-forge/gz-sim-feedstock/pull/19 I backported some relocatibility fixes, that permit us to test how Windows works with the latest verson of Gazebo Garden.

## gz sim -s 

`gz sim -s` works fine, except for a few warnings related to logs and env variables:

~~~
(gz-sim7) C:\Users\STraversaro\AppData\Local\mambaforge\envs\gz-sim7\Library\lib\ruby\gz>gz sim -s --verbose
[Wrn] [D:\bld\gz-common5_1686049407341\work\src\Filesystem.cc:50] Failed common::createDirectories (ec: system:123 The filename, directory name, or volume label syntax is incorrect.)
Error opening log file: C:\Users\STraversaro\.gz\sim\log\2023-06-21T09:40:20.288175600\server_console.log
[Msg] Gazebo Sim Server v7.5.0
[Msg] Loading default world.
[Msg] Serving entity system service on [/entity/system/add]
[Msg] Loaded level [3]
[Msg] No systems loaded from SDF, loading defaults
[Wrn] [D:\bld\gz-common5_1686049407341\work\src\Util.cc:380] Reading environment variable with _allowEmpty==true is unsupported on Windows.
[Wrn] [D:\bld\gz-common5_1686049407341\work\src\Util.cc:380] Reading environment variable with _allowEmpty==true is unsupported on Windows.
[Msg] Create service on [/world/default/create]
[Msg] Remove service on [/world/default/remove]
[Msg] Pose service on [/world/default/set_pose]
[Msg] Pose service on [/world/default/set_pose_vector]
~~~

## gz sim -g 

gz sim -g fails as it does not find some gz-gui plugins:

~~~
(gz-sim7) C:\Users\STraversaro>gz sim -g
[GUI] [Err] [D:\bld\gz-gui7_1686132745745\work\src\Application.cc:487] Failed to load plugin [MinimalScene] : couldn't find shared library.
[GUI] [Err] [D:\bld\gz-gui7_1686132745745\work\src\Application.cc:487] Failed to load plugin [InteractiveViewControl] : couldn't find shared library.
[GUI] [Err] [D:\bld\gz-gui7_1686132745745\work\src\Application.cc:487] Failed to load plugin [CameraTracking] : couldn't find shared library.
[GUI] [Err] [D:\bld\gz-gui7_1686132745745\work\src\Application.cc:487] Failed to load plugin [MarkerManager] : couldn't find shared library.
[GUI] [Err] [D:\bld\gz-gui7_1686132745745\work\src\Application.cc:487] Failed to load plugin [WorldControl] : couldn't find shared library.
[GUI] [Err] [D:\bld\gz-gui7_1686132745745\work\src\Application.cc:487] Failed to load plugin [WorldStats] : couldn't find shared library.
[GUI] [Err] [D:\bld\gz-gui7_1686132745745\work\src\Application.cc:487] Failed to load plugin [Screenshot] : couldn't find shared library.
~~~

The problem here is that I forgot to propagate the relocatibility fixes in gz-gui, but it should be trivial to do so.

## gz sim

gz sim fails as there is no `fork` on Windows:

~~~
(gz-sim7) C:\Users\STraversaro\AppData\Local\mambaforge\envs\gz-sim7\Library\lib\ruby\gz>gz sim
C:/Users/STraversaro/AppData/Local/mambaforge/envs/gz-sim7/Library/lib/ruby/gz/cmdsim7.rb:503:in `fork': fork() function is unimplemented on this machine (NotImplementedError)
        from C:/Users/STraversaro/AppData/Local/mambaforge/envs/gz-sim7/Library/lib/ruby/gz/cmdsim7.rb:503:in `execute'
        from C:/Users/STraversaro/AppData/Local/mambaforge/envs/gz-sim7/Library/bin//gz:308:in `<main>'
~~~

traversaro, 2023-07-04:
I backported more relocatability fixes, and now `gz sim -s --verbose` is starting correctly, even with some warnings:

~~~
(gz-sim) C:\src\gzws2023\gz-rendering\examples\simple_demo_qml\build>.\simple_demo_qml.exe
[Msg] Loading plugin [gz-rendering-ogre2]
[Err] [C:\src\gzws2023\gz-rendering\ogre2\src\Ogre2RenderEngine.cc:599] Unable to find Ogre Plugin[C:\Users\STraversaro\AppData\Local\mambaforge\envs\gz-sim\Library\lib\OGRE-Next\RenderSystem_GL3Plus]. Rendering will not be possible.Make sure you have installed OGRE properly.
[Err] [C:\src\gzws2023\gz-rendering\ogre2\src\Ogre2RenderEngine.cc:1131]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): wglMakeCurrent failed: The pixel format is invalid.
 in Win32Window::create at C:\src\gzws2023\ogre-next\RenderSystems\GL3Plus\src\windowing\win32\OgreWin32Window.cpp (line 624)
[Err] [C:\src\gzws2023\gz-rendering\ogre2\src\Ogre2RenderEngine.cc:1131]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): currentGLContext was specified with no current GL context in Win32Window::create at C:\src\gzws2023\ogre-next\RenderSystems\GL3Plus\src\windowing\win32\OgreWin32Window.cpp (line 230)
~~~

On the other hand, after patching `cmdgzsim7.rb` to permit `gz sim -g` even on Windows,  `gz sim -g --verbose` still fails with "wglMakeCurrent failed: The pixel format is invalid." error:
~~~
(gz-sim) C:\Users\STraversaro>gz sim -g --verbose
[Msg] Gazebo Sim GUI    v7.5.0
[GUI] [Msg] Loading config [C:\Users\STraversaro\.gz\sim\7\gui.config]
[GUI] [Msg] Added plugin [3D View] to main window
[GUI] [Msg] Loaded plugin [MinimalScene] from path [C:\Users\STraversaro\AppData\Local\mambaforge\envs\gz-sim\Library\lib\gz-gui-7\plugins\MinimalScene.dll]
[GUI] [Wrn] [D:\bld\gz-gui7_1688332124286\work\src\Application.cc:860] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:52:3: QML RenderWindowOverlay: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Wrn] [D:\bld\gz-gui7_1688332124286\work\src\Application.cc:860] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:67:3: QML EntityContextMenu: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Wrn] [D:\bld\gz-gui7_1688332124286\work\src\Application.cc:860] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:52:3: QML RenderWindowOverlay: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Wrn] [D:\bld\gz-gui7_1688332124286\work\src\Application.cc:860] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:67:3: QML EntityContextMenu: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Wrn] [D:\bld\gz-gui7_1688332124286\work\src\Application.cc:860] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:52:3: QML RenderWindowOverlay: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Wrn] [D:\bld\gz-gui7_1688332124286\work\src\Application.cc:860] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:67:3: QML EntityContextMenu: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Wrn] [D:\bld\gz-gui7_1688332124286\work\src\Application.cc:860] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:52:3: QML RenderWindowOverlay: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Wrn] [D:\bld\gz-gui7_1688332124286\work\src\Application.cc:860] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:67:3: QML EntityContextMenu: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Wrn] [D:\bld\gz-gui7_1688332124286\work\src\Application.cc:860] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:52:3: QML RenderWindowOverlay: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Wrn] [D:\bld\gz-gui7_1688332124286\work\src\Application.cc:860] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:67:3: QML EntityContextMenu: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Wrn] [D:\bld\gz-gui7_1688332124286\work\src\Application.cc:860] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:52:3: QML RenderWindowOverlay: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Wrn] [D:\bld\gz-gui7_1688332124286\work\src\Application.cc:860] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:67:3: QML EntityContextMenu: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Msg] Added plugin [Entity Context Menu] to main window
[GUI] [Msg] Loaded plugin [EntityContextMenuPlugin] from path [C:\Users\STraversaro\AppData\Local\mambaforge\envs\gz-sim\Library\lib\gz-sim-7\plugins\gui\EntityContextMenuPlugin.dll]
[GUI] [Msg] Added plugin [Scene Manager] to main window
[... cut ...]
xerces-c                  3.2.4                h63175ca_2    conda-forge
xz                        5.2.6                h8d14728_0    conda-forge
yaml                      0.2.5                h8ffe710_2    conda-forge
zeromq                    4.3.4                h0e60522_1    conda-forge
zlib                      1.2.13               hcfcfb64_5    conda-forge
zstd                      1.5.2                h12be248_6    conda-forge
zziplib                   0.13.69              h1d00b33_1    conda-forge
~~~

traversaro, 2024-03-22:
For anyone reaching this message after trying to launching `gz sim` on Windows. Indeed, as of March 2024 the gz sim with gui is not working on Windows, but a workaround for now is to use `gz sim` from WSL, in the past it had some problems, but now all problems should have been solved and it should run fine for the latest version of Gazebo Garden and Harmonic, see https://community.gazebosim.org/t/wslg-with-gpu-support-available-on-latest-version-of-gazebo-garden-and-harmonic/2360 for more details.

fyi @xela-95

ChristianColetti, 2025-04-24:
Has development of gz-sim for windows without WSL stopped?

traversaro, 2025-04-26:
> Has development of gz-sim for windows without WSL stopped?

The answer to this question I think it depends on your definition of "development" or "stopped". Anyhow, the PR https://github.com/gazebosim/gz-sim/pull/2849 will finally provide a `gz sim` command that should work on native Windows without the need to run separately `gz sim -s` and `gz sim -g`, that I think would be a good step forward into support gz-sim for Windows without WSL.

j-rivero, 2025-06-11:
> * Fix failing tests (153 failures of 349 tests)

Revisiting this old time classic, I'm going to call i done the fix of the failing test since we have a long months without failures in Windows. We have disabled a good bunch of them but most of them are now functional.

For the records, this is the URL for seeing the more than 1100 tests run on Windows for gz-sim 9 https://build.osrfoundation.org/view/gz-ionic/job/gz_sim-9-clowin/lastCompletedBuild/testReport/(root)/

Foadsf, 2026-09-07:
Some measured data from a native Windows machine, since this thread is long and the newest status in it is from early 2024.

**Setup**: Windows 11 Enterprise 10.0.26200, gz-sim **10.5.0** installed exactly per the [Windows binary tutorial](https://gazebosim.org/docs/latest/install_windows/) (pixi + conda-forge). NVIDIA RTX A2000 + Intel UHD, physical console session.

**Headless works.** `gz sim -s -v4 -r --iterations 200 shapes.sdf` loads DART, serves the scene topics and shuts down cleanly. No manual `server.config` copying was needed — the relocatability problems discussed earlier in this thread appear to be resolved.

**The GUI does not.** `gz sim -v4 shapes.sdf` — the command the tutorial presents as the working one — never opens a window; both processes stay resident and responsive with no window mapped. Filed separately as #3960. The throw is in `RenderEngineManager` → `SystemPaths::FindSharedLibrary`, it is engine-independent (`--render-engine ogre` fails identically), and it happens before any GL context exists, so it is **not** the `wglMakeCurrent` / pixel-format failure from #2089.

**One thing this thread can drop.** The `DynamicFactory(): Unable to parse descriptor set from [...gz-msgs12.gz_desc]` line is a Windows text-mode read of a binary file. The installed file contains `0x1A` 590 times, first at offset 146 of 84,698 — so a text-mode stream stops at byte 146. Released `gz-msgs12_12.0.2` uses `std::ifstream ifs(descFile);`; the tip of `gz-msgs12` already uses `std::ios::binary`. That one resolves itself with the next gz-msgs release.

**Not everything is broken.** gz-math builds with MSVC and passes **130/130 tests** on Windows today, both locally and on a `windows-2022` runner.

**On CI, which is what keeps this from recurring.** Neither gz-sim nor gz-math has a Windows job in GitHub Actions. Windows is exercised only on the buildfarm and in the conda-forge feedstocks, both after the fact, so a Windows break is invisible on the PR that causes it. Two PRs, each verified green on a fork before submitting:

- gazebosim/gz-sim#3961 — builds the source tree with MSVC against the conda-forge dependency set, 20 min cold-cache
- gazebosim/gz-math#843 — build **plus** the full test suite, 130/130

---

**Docs request.** Could someone update <https://gazebosim.org/docs/all/getstarted/>? It currently says only *"Support via Conda-Forge is not fully functional, as there are known runtime issues"*, and the install table has no Windows row at all. That undersells the real state in both directions: headless simulation genuinely works on native Windows now, and the page does not link the Windows install pages that do exist ([binary](https://gazebosim.org/docs/latest/install_windows/), [source](https://gazebosim.org/docs/latest/install_windows_src/)).

A Windows row with an honest scope note — something like *"headless supported; GUI experimental, see #3960"* — would stop readers concluding from this page that WSL2 is the only option, which is the impression both it and this thread currently give.

micmzr, 2026-09-15:
The GUI launches correctly for the command `gz sim -v4 shapes.sdf`. 
Tested on robostack:rolling.