gazebosim/gz-sim #177: Show selected entity when opening Component Inspector

nkoenig, 2020-06-03:
The GUI Component Inspector plugin always starts with the World properties shown. If I have an entity selected, then I would like the Component Inspector to open with the properties of the selected entity displayed.

chapulina, 2020-06-08:
The main problem implementing this is that there's no central place keeping the ground truth about what entities are selected. The 3D scene only knows about the entities with a visual, the entity tree knows about them all but that plugin is optional...

For this reason, we've been purposely deselecting all entities when a plugin that relies on selections to work is loaded. This way we can be sure that all selections are caught by the plugin.

So how about this alternative: adding an item to the entity's context menu saying "Inspect", which loads the inspector for that entity? I think it should accomplish what you're trying to achieve.

naitikpahwa18, 2026-01-22:
The reasoning around selection state and intentional deselection is clear. Given those constraints, an “Inspect” option in the entity context menu still seems like a sensible approach.

@chapulina If this aligns with the current GUI direction, I’d like to work on implementing it.