gazebosim/gz-sim #188: Create joints from the GUI

chapulina, 2020-06-05:
Gazebo-classic has an interface to create joints in edit mode. See the [Model Editor tutorial](http://gazebosim.org/tutorials?tut=model_editor#CreateJoints). We should port this to Ignition.

[image]

Key features:

* Pick child and parent links either from a dropdown or clicking on the 3D scene
* Configure all joint properties on the dialog (on Ignition this could be done in the component inspector)
* Joint is visualized as a line and an axis (requires #106 )

peci1, 2022-02-01:
I think this has already been done in the recent Model Editor PR.

chapulina, 2022-03-01:
True, I'm closing this, but we still should address #1232 to wrap up the model editor work.

closed 2022-03-01 by chapulina

mkschreder, 2023-02-09:
Is this supposed to be available in gz-garden?