gazebosim/gz-sim #190: Insert system plugins from the GUI

chapulina, 2020-06-05:
Often users forget to load a necessary plugin into simulation. It would be handy if they could pick plugins from a menu or a browser window and attach them to entities at runtime.

The backend already supports loading systems at runtime. What's missing:

* A menu / browser window to choose system plugins
* A dialog to add configurations to the plugin (including what entity to attach to)
* A transport interface for the GUI to tell the server to load a new system

As a reference, Gazebo-classic's model editor supported adding model plugins:

[image]

---

It's worth mentioning that GUI plugins can already be loaded from the top-right menu.

mayman99, 2022-02-23:
If no one is working on this already, I could have a look at this and maybe at #1258 after.

chapulina, 2022-03-01:
Thanks, @mayman99 ! Before starting the implementation, it would be good to draw a high-level schematic of how the plugin could work so we can iterate on that before writing any code. 

I think it would also be good to start with a PR that adds just the transport interface, and subsequent PRs add the GUI.

mayman99, 2022-03-06:
Thanks! @chapulina .
I'll be having a closer look at how the loading of plugins from the menu on top right works. The thing I so far don't have a concrete idea about is the transport interface, so if the backend already supports loading systems at runtime, why do we need a transport interface then? (I understand what the `UserCommands` class do, and that a plugin must inherit from it's plugin interface/interfaces-- but still having trouble connecting all together).

chapulina, 2022-03-09:
> if the backend already supports loading systems at runtime, why do we need a transport interface then? 

Currently, systems can be loaded at runtime if they're attached to a model that's being spawned. But the idea here is to add a system to a model / sensor / visual which is already in simulation, or to the world itself. We don't have an interface for that yet. 

I think the `SimulationRunner` could offer a transport service that accepts a message that contains `msgs::Plugin` + `msgs::Entity` and calls `AddSystem` accordingly.

iche033, 2022-06-17:
Here's my draft mock-up of what I have in mind for the GUI front end. We can add a button / icon to the Component inspector inside `System Plugin Info` for adding new systems:

[image]

Once the user click on the button / icon, a window pops up like in gazebo-classic and let user fill out the system plugin details.

In the future we can add a trash can icon next to each System for removing them.

How does that look?

chapulina, 2022-06-17:
How about putting an icon on the top toolbar like we have on the `ComponentInspectorEditor` for links?

https://github.com/gazebosim/gz-sim/pull/1251/files#diff-fa267096dafbb2e7c4e4c9d9566bfb2537da09dac37d11e98b38e1c4d668d2bb

[image]

We could create a simple icon that looks like a plug, for plugin 😄

iche033, 2022-06-17:
> How about putting an icon on the top toolbar like we have on the ComponentInspectorEditor for links?

sure that works. I'll give that a try

chapulina, 2022-08-03:
* https://github.com/gazebosim/gz-sim/pull/1549

closed 2022-08-03 by chapulina