gazebosim/gz-sim #195: Scaling tool

chapulina, 2020-06-06:
The [Transform Control tool](https://ignitionrobotics.org/docs/citadel/manipulating_models#transform-control) lets users translate and rotate entities in simulation. We should also add a scaling tool to that widget.

[image]

Ignition Rendering already supports scaling all types of visuals. The main missing piece right now is physics, see https://github.com/ignitionrobotics/ign-physics/issues/65. Once that issue is solved, we can:

* Add a scaling command to [UserCommands](https://github.com/ignitionrobotics/ign-gazebo/tree/master/src/systems/user_commands)
* Add a scaling button to [TransformControl](https://github.com/ignitionrobotics/ign-gazebo/tree/master/src/gui/plugins/transform_control)
* We may need to add scaling "heads" to Ignition Rendering's [AxisVisual](https://github.com/ignitionrobotics/ign-rendering/blob/master/include/ignition/rendering/AxisVisual.hh) or some other visual

kulkarni-raunak, 2024-06-06:
Are there any updates on this?

kulkarni-raunak, 2024-06-06:
Is this scaling option available Gazebo Garden? If so, how to achieve it?

javier99nl, 2024-12-04:
Hello, I have a problem when scaling models from the sdf such as:
<include>
  <uri>model://water_tower</uri>
  <pose>-11 -7 -12 0 0 0</pose>
  <scale>0.001 0.001 0.001</scale>
</include>

This is not really working

Dafodilrat, 2025-04-17:
Any updates on this feature ??