gazebosim/gz-sim #201: Uniform the APIs to apply forces and wrenches to a link

diegoferigo, 2020-06-09:
The application of link forces and wrenches is performed by the following two methods:

https://github.com/ignitionrobotics/ign-gazebo/blob/d7cd60a7ec8d5931f54d155894b66974dbcf78a2/include/ignition/gazebo/Link.hh#L187-L192

https://github.com/ignitionrobotics/ign-gazebo/blob/d7cd60a7ec8d5931f54d155894b66974dbcf78a2/include/ignition/gazebo/Link.hh#L194-L202

As you can notice, the former applies a force to the CoM of the link, the latter to the link origin. This mismatch could be somehow misleading.

Partially related, it would be also nice having a method to apply a wrench for more than one physics step. In the case of a small physics step size and multiple iterations, a single step could be sometimes limiting. Though, I think that it's not that straightforward and it would require both a new component and some modification to the Physics system.

chapulina, 2020-06-10:
This feels like deja vu :sweat_smile: https://github.com/osrf/gazebo/issues/354

diegoferigo, 2020-06-10:
Ahaha you have a good memory @chapulina! :smile: The match is impressing, not one but two same requests in a single issue, wow. Not sure if I should feel disturbed by this xD