gazebosim/gz-sim #203: Reset models and worlds similarly to Ctrl + R in Gazebo Classic

diegoferigo, 2020-06-10:
I see a lot of new GUI-related issues to track new features, and that's great! There's one feature, though, that in Gazebo Classic was used extensively by pretty much all the people I know that's still missing: resetting a world with `Ctrl + R`.

Gazebo Classic had the following APIs:

- [`gazebo::physics::Model::Reset`](https://osrf-distributions.s3.amazonaws.com/gazebo/api/11.0.0/classgazebo_1_1physics_1_1Model.html#af82f4acaed7bb39d568e689a9caa63d5)
- [`gazebo::physics::World::Reset`](https://osrf-distributions.s3.amazonaws.com/gazebo/api/11.0.0/classgazebo_1_1physics_1_1World.html#a372de693ad40b3f42839c8ec6ac845f4)

I never had a look at how it was implemented previously, what I want to add to this issue as a reminder is that it would be nice having a sort of callback that world and model plugins could use to reset their state. 

A simple alternative would be adding a new reset component and ask plugin developers to always check if this component is present and take action. Both would equally work, at first thought I don't see major limitations of any of them. Maybe the callback would propagate better and more clearly.

Partially related to #113.

diegoferigo, 2020-06-11:
Great, thanks for providing these details! If I understood well, there's already the initial support for something similar than Ctrl+R, actually more advanced since it allows selecting `t=t_k` instead of `t=0` by default. However, so far there's no System yet (including Physics) that use this feature.

Few comments:

- While it is handy having a transport-based interface, a C++ API is still missing. I don't want to go much in detail into design choices (discussions about network transparency probably would require a separated issue) but in my opinion one of the selling points of Ignition Gazebo is the possibility of using it as a library without the need to use transport. In past few years other simulators like mujoco and pybullet got an remarkable user-base thanks to this feature and I hope that also Ignition Gazebo could reach a similar feature-set (and way beyond :)).
- Beyond time seeking, I still think that an API like `World::Reset` and `Model::Reset` quite useful.
- Regarding the reset of the Physics, I have the feeling that rolling back time (by issuing negative time steps) and resetting the state of the physics engine + aligning the ECM are two very different approaches that could be implemented separately. The former could involve a (backward?) integration, the latter just a reset of the state taking care of all the hidden states like integrator gains in the case of PIDs (maybe the reset components of [this PR](https://osrf-migration.github.io/ignition-gh-pages/#!/ignitionrobotics/ign-gazebo/pull-requests/437/page/1) could help). Using time = 0 maybe it's easier since the state could be gathered from the SDF of models and world, a reset to a given time would require the logger being enabled.

chapulina, 2020-06-11:
> so far there's no System yet (including Physics) that use this feature.

That's true for live simulations, but the `LogPlayback` system supports it when playing back log files.

> a C++ API is still missing

That's a valid point. Since time management happens outside of the ECM, I think a component approach would be tough to implement though. Probably an event like we have for pause would be more appropriate:

https://github.com/ignitionrobotics/ign-gazebo/blob/d7cd60a7ec8d5931f54d155894b66974dbcf78a2/src/SimulationRunner.cc#L103-L104

> rolling back time (by issuing negative time steps) and resetting the state of the physics engine + aligning the ECM are two very different approaches that could be implemented separately

Fully agreed. The message interface already keeps those separate, with reset / rewind separate from seek.

diegoferigo, 2021-10-18:
xref https://github.com/ignitionrobotics/ign-gazebo/issues/1107