gazebosim/gz-sim #215: Build the whole Ignition stack as a single cmake project

joxoby, 2020-06-19:
I'm curious as to why there's no support to build the whole stack as a single cmake project, something like:
```
cmake_minimum_required(VERSION 3.10.2)

project(ignition VERSION 1.0.0)

add_subdirectory(ign-cmake)
add_subdirectory(ign-math)
add_subdirectory(ign-physics)
...
```
I would prefer to not have to use `colcon` and have clear separation of `config`, `build` and `install` stages. I understand that I can do this for each project individually, but it would be nice if the previous structure was supported.

chapulina, 2020-06-19:
Would there be a way to support this approach while still allowing each package to be compiled individually? As long as each library can still be used on its own, we could consider also supporting this use case.

traversaro, 2020-06-20:
> 
> 
> Would there be a way to support this approach while still allowing each package to be compiled individually? As long as each library can still be used on its own, we could consider also supporting this use case.

I maintain some internal projects that use [`FetchContent`](https://cmake.org/cmake/help/v3.11/module/FetchContent.html) + `add_subdirectory` to build together several libraries, and indeed is an extremely convenient workflow, for example for cross-compilation. One possible strategy to deal with this is to avoid calling `find_package` directly for dependencies that may be imported with `FetchContent`, but rather  first check if the relative imported target is  already available, for example for `Eigen3`: 
~~~
if(NOT TARGET Eigen3::Eigen)
  find_package(Eigen3 CONFIG)
endif()
~~~
see https://github.com/robotology/idyntree/blob/devel/cmake/iDynTreeDependencies.cmake#L56 and https://github.com/robotology/idyntree/blob/devel/cmake/iDynTreeDependencies.cmake#L14 for an example of such use. 

As most of ignition projects already use [`ign_find_package`](https://github.com/ignitionrobotics/ign-cmake/blob/ignition-cmake2_2.2.0/cmake/IgnUtils.cmake#L136) to find their dependencies, a first step  would be  to modify it to have a `MAIN_TARGET`  option, that skips the `find_package` call if the target is already defined. Unfortunately, then  it would be necessary to modify the `ign_find_package`  invocations for dependencies that you want to support via `add_subdirectory` and add the `MAIN_TARGET` option. If some projects do not export targets (I am not sure if `ignition-cmake` exports them) some other solution needs to be found.

traversaro, 2020-06-20:
This modification would also permit to consume the Ignition libraries using [`CPM`](https://github.com/TheLartians/CPM.cmake).

traversaro, 2020-06-20:
> As most of ignition projects already use [`ign_find_package`](https://github.com/ignitionrobotics/ign-cmake/blob/ignition-cmake2_2.2.0/cmake/IgnUtils.cmake#L136) to find their dependencies, a first [...]

Actually, at least for ignition projects that provide a  consistent naming scheme for CMake imported targets, some logic  for determining automatically the `MAIN_TARGET` could  be present in `ign_find_package` itself, removing the need to modify all the Ignition Libraries CMakeLists.txt files. `MAIN_TARGET` may still be necessary if other non-ignition dependencies want to be supported via `FetchContent`/`add_subdirectory`.

joxoby, 2020-06-22:
This is a very interesting idea @traversaro. What's the best way to deal with target name collisions? For example, every ignition library calls `ign_setup_target_for_codecheck` that defines a target `codecheck`, which will collide with the one defined by other libraries.

traversaro, 2020-06-22:
> This is a very interesting idea @traversaro. What's the best way to deal with target name collisions? For example, every ignition library calls `ign_setup_target_for_codecheck` that defines a target `codecheck`, which will collide with the one defined by other libraries.

It depends, for some targets that are inherently generic, such as `uninstall`, for which it is sufficient that one projects adds it and then it applies for all the super-project, it is  sufficient to add a condition check like `if(NOT TARGET uninstall)`, see https://gitlab.kitware.com/cmake/community/-/wikis/FAQ#can-i-do-make-uninstall-with-cmake .

For more development-related targets, it could make sense to have a generic `ignition-cmake` option to control if generate or not this dev targets. In this way, you can set this option in the CMake code to avoid generating these targets: 

~~~cmake
cmake_minimum_required(VERSION 3.10.2)

project(project-that-uses-ignition VERSION 1.0.0)

set(IGNITION_CREATE_DEV_TOOLS OFF CACHE BOOL "" FORCE)
add_subdirectory(ign-cmake)
add_subdirectory(ign-math)
add_subdirectory(ign-physics)
...
~~~

A common workflow is to change appropriately the default value of these options depending of whether a project is built as standalone or is added via `add_subdirectory`. This is typically done by checking if `CMAKE_CURRENT_SOURCE_DIR` is equal to `CMAKE_SOURCE_DIR`, see for example https://github.com/gabime/spdlog/blob/1b14fa53efc7423ff604afc1249c74fd9aefadf3/CMakeLists.txt#L51 and https://github.com/fmtlib/fmt/blob/b287c37c651d54669032e3e4685b4423409e4d00/CMakeLists.txt#L12 .