gazebosim/gz-sim #217: unable to install using the binaries

jhwangbo, 2020-06-20:
on linux 18.04, I get the following error when following the installation instruction (https://ignitionrobotics.org/docs/acropolis/install#option-1-installation-on-ubuntu-bionic)

`Unpacking ignition-gazebo (1.1.0-1~bionic) ...
dpkg: error processing archive /tmp/apt-dpkg-install-uGiRNL/338-ignition-gazebo_1.1.0-1~bionic_amd64.deb (--unpack):
 trying to overwrite '/usr/bin/ign-gazebo', which is also in package ignition-gazebo2 2.20.1-1~bionic`

I tried it on two different computers and I get the same result. any fix to this issue?

traversaro, 2020-06-20:
Note that, as stated in the top of the page:
> Acropolis has reached end-of-life on September 2019. This page exists for archival purposes. Please refer to the latest supported version.

To have the most recent install version of ignition-gazebo, you can follow the instructions at: 
https://ignitionrobotics.org/docs/citadel/install_ubuntu 

However, Ignition Citadel will install Ignition Gazebo 3, while it seems that you already have ignition-gazebo2 installed, so perhaps you already installed the ignition blueprint release  that contained `ignition-gazebo2`. If you do not need it anymore, you can uninstall it with the  following  command:
~~~
sudo apt remove ignition-gazebo2
~~~

jhwangbo, 2020-06-21:
Then it seems like a bad link/documentation. I went to the Ignition robotics website, then doc -> get started -> the latest install tutorial.

chapulina, 2020-06-22:
I'll close this in favour of https://github.com/ignitionrobotics/docs/issues/4

closed 2020-06-22 by chapulina