gazebosim/gz-sim #220: Failing tests in citadel branches

joxoby, 2020-06-24:
I periodically pull from the ignition citadel branches to get some new features that I need. I noticed that some tests keep failing, not only on my system but on CI too (for example, https://github.com/ignitionrobotics/ign-gazebo/runs/758657719). Are there any particular reasons as to why code that break the tests gets in?

chapulina, 2020-06-24:
The GitHub actions CI is still under development, so we haven't been paying attention to it.

The CI that we care about at the moment is on Jenkins. For Citadel:

Ubuntu Bionic: [image]
OSX: [image]

There are a few known failing tests on OSX which are being disabled on #209.

joxoby, 2020-06-24:
As always, thanks for the quick responses @chapulina.

chapulina, 2020-06-24:
No worries. I'll leave this open as a reminder to fix GitHub actions :smile: Related to https://github.com/ignition-tooling/ubuntu-bionic-ci-action/issues/6.

chapulina, 2020-10-05:
The GitHub actions CI is running successfully for most branches, with the exception of some flaky tests. I think this can be closed.

* https://github.com/ignitionrobotics/ign-gazebo/actions?query=workflow%3A%22Ubuntu+CI%22+branch%3Aign-gazebo2
* https://github.com/ignitionrobotics/ign-gazebo/actions?query=workflow%3A%22Ubuntu+CI%22+branch%3Aign-gazebo3
* https://github.com/ignitionrobotics/ign-gazebo/actions?query=workflow%3A%22Ubuntu+CI%22+branch%3Amain

closed 2020-10-05 by chapulina