gazebosim/gz-sim #226: <joint><axis><limit><effort> not enforced

caguero, 2020-06-25:
Modify the `examples/worlds/apply_joint_force.sdf` adding an effort tag:

```
<joint name="j1" type="revolute">
        <pose>0 0 -0.5 0 0 0</pose>
        <parent>base_link</parent>
        <child>rotor</child>
        <axis>
          <xyz>0 0 1</xyz>
          <dynamics>
            <damping>0.0001</damping>
          </dynamics>
          <limit>
            <effort>0.0001</effort>
          </limit>
        </axis>
      </joint>
```

Launch Gazebo:

```
ign gazebo -v 4 ../examples/worlds/apply_joint_force.sdf
```

Apply the theoretical max torque:

```
ign topic -t "/model/joint_force_example/joint/j1/cmd_force" -m ignition.msgs.Double  -p "data: 0.0001"
```

Observe how the joint starts spinning relatively slow.

Now, apply a bigger torque:

```
ign topic -t "/model/joint_force_example/joint/j1/cmd_force" -m ignition.msgs.Double  -p "data: 0.001"
```

The joint will start spinning faster than before but it shouldn't, as the first command was already applying the maximum allowed torque.

chapulina, 2020-06-25:
This sounds similar to #214 , but specific to `<effort>`.

caguero, 2020-06-25:
True. Not sure if all elements of `<limit>` are ignored. If that's the case, we can close this ticket in favor of #214 .

azeey, 2020-06-25:
I believe the issue #214 occurs only when using `components::JointVelocityCmd`, which is used by the JointController system. Fixing that will take some time since it requires a change in DART. The issue here is something we can fix in ign-physics, so it might be worth keeping them separate.

azeey, 2020-07-01:
Fixed by https://github.com/ignitionrobotics/ign-physics/pull/74

closed 2020-07-01 by azeey