gazebosim/gz-sim #231: 3D plot plugin

chapulina, 2020-07-02:
Gazebo-classic's [LinkPlot3DPlugin](https://github.com/osrf/gazebo/blob/gazebo11/plugins/LinkPlot3DPlugin.hh) traces the trajectory of a link in the rendering scene by publishing 3D markers.

Try `gazebo worlds/plot3d.world `:

[image]

It would be nice to port it to Ignition. We already have markers, so it should be a matter of implementing a system plugin that publishes markers according to the current link poses.

luccosta, 2020-08-04:
Hi @chapulina, I'm interested on this Issue. I'm planning to create a file with for the plugin as in Gazebo-classic. Should I put this on `/src`? And what branch to target?

> We already have markers

Planning to use [marker example](https://github.com/ignitionrobotics/ign-gazebo/blob/master/examples/standalone/marker/marker.cc) as guide.

chapulina, 2020-08-04:
Thanks, @luccosta !

You can create a new directory under [src/systems](https://github.com/ignitionrobotics/ign-gazebo/tree/master/src/systems) for your new system. I think you'll probably use some `Each` calls on a `PostUpdate` callback to get link poses and publish the marker messages.

The markers are available since `ign-gazebo3`, which is Ignition Citadel, so it would be great if you targeted that branch and after merging we'll forward-port it to `master`.

chapulina, 2021-07-30:
Done in #917

PR #917 (2021-07-29): 3D plot GUI plugin
# 🎉 New feature

Closes #231 

Alternative to #332

## Summary

Plots the path of entities in the 3D scene, parameters are customizable from SDF and / or the GUI.

~~WIP while I work on some final usability tweaks, and possibly a test.~~

Ready for review!

## Test it

```
ign gazebo -r plot_3d.sdf
```

This is what it currently looks like:

[image]

## Checklist
- [x] Signed all commits for DCO
- [ ] Added tests
- [x] Added example and/or tutorial
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] `codecheck` passed (See [contributing](h [... cut]
closed 2021-07-30 by chapulina