gazebosim/gz-sim #235: Support DEMs

caguero, 2020-07-08:
Once heightmap support is ready, we should add support to load a Digital Elevation Map. When loaded, we need to provide a basic API to query the main properties of the DEM, and then, load it as a heightmap.

chapulina, 2022-03-03:
Here's the latest series of PRs:

* https://github.com/ignitionrobotics/ign-common/pull/292
* https://github.com/ignitionrobotics/ign-rendering/pull/539
* https://github.com/ignitionrobotics/ign-rendering/pull/560
* https://github.com/ignitionrobotics/ign-physics/pull/315
* https://github.com/ignitionrobotics/ign-gazebo/pull/1262

jennuine, 2022-03-24:
All PRs are merged.

closed 2022-03-24 by jennuine

Ryanf55, 2023-02-01:
Quick question, is there any documentation or tutorials on how to use this feature? I was unable to find any.

jennuine, 2023-02-01:
Unfortunately, the only tutorial available is from gazebo-classic found here: https://classic.gazebosim.org/tutorials?tut=dem&cat=build_world

It is outdated but will hopefully get you in the right direction. When following the tutorial, you should be able to copy this section into a Gazebo Garden compatible `SDF` world: https://github.com/gazebosim/gazebo-classic/blob/1bef2dc573aa15ca8a56b4a7dadb9905cf978582/worlds/heightmap_dem.world#L8-L54

If you'd like to contribute, feel free to open a PR for a newer tutorial on DEMs here: https://github.com/gazebosim/docs :smile: