gazebosim/gz-sim #239: Name component serialization doesn't work for names with spaces.

luca-della-vedova, 2020-07-14:
I have been playing with some models that have spaces in their names, such as [Pine Tree](https://app.ignitionrobotics.org/OpenRobotics/fuel/models/Pine%20Tree) and noticed that when I spawn multiple models in the same world, such as:
```
    <include>
      <name>Pine Tree</name>
      <uri>model://Pine Tree</uri>
      <pose>78.16778723671018 -35.70987539227364 0 0 0 1.5708</pose>
      <static>True</static>
    </include>
    <include>
      <name>Pine Tree_2</name>
      <uri>model://Pine Tree</uri>
      <pose>74.0254778248307 -32.28629661564661 0 0 0 1.5708</pose>
      <static>True</static>
    </include>
```
I get the following error (and the model is spawned black):
```
[ign-8] [GUI] [Err] [SceneManager.cc:159] Visual: [Pine] already exists
```
I tracked down the problem being in the Name component missing the second word, and I believe this is due to the way we are doing serialization and deserialization in the Component template, specifically if I understand correctly we use the operator << for serialization and the operator >> for deserialization.
However, this doesn't work greatly for strings since, in the case of strings with spaces, the >> operator will only operate until the space (hence both models are seen as having the name "Pine").
By contrast this world file works OK for gazebo classic, do you think it should be fixed or should we just impose a "No model names with spaces" policy?

chapulina, 2020-07-14:
> this world file works OK for gazebo classic

If I recall correctly, classic's support for names with spaces is spotty. I could find a couple issues related to using spaces in resources:

* https://github.com/osrf/gazebo/issues/2056
* https://github.com/osrf/gazebo/issues/2369

> this is due to the way we are doing serialization and deserialization in the Component template

Thank you for tracking down the source of the problem. Regardless of whether we want to support names with spaces, this sounds like a bug on string components. We can create a custom serializer for `std::string` that doesn't use operators. I'll work on that :ok_hand: 

> should we just impose a "No model names with spaces" policy?

Once the serialization issue is solved, I imagine we may run into others, such as [topic names](https://ignitionrobotics.org/api/transport/9.0/nodestopics.html).

I don't think `components::Name` should be opinionated about this kind of detail though. There will be use cases where users would like to use pretty human-readable names for their entities and it won't break anything in their setup. So I think this could be handled at each system's level. For example, systems that use entity names for topics could convert `Pine Tree` to `pine_tree` and document that.

chapulina, 2020-07-15:
Added a string serializer here, meshes and models seem to work fine: https://github.com/ignitionrobotics/ign-gazebo/pull/244

The next step is going over all topics that use entity names.

chapulina, 2020-10-05:
Related to this, we may want to replace `::` in entity names with `/` on topics. See https://github.com/ignitionrobotics/ros_ign/issues/114

closed 2021-01-26 by chapulina