gazebosim/gz-sim #240: Support object populations

caguero, 2020-07-14:
The idea is to be able to specify a population of objects in SDF with some parameters (e.g.: distribution type, num objects).

Here's a reference pull request:

https://osrf-migration.github.io/gazebo-gh-pages/#!/osrf/gazebo/pull-requests/1157/page/1

And a reference tutorial:

http://gazebosim.org/tutorials?tut=model_population&cat=build_world

TanguyGerniers, 2023-06-20:
Hi @caguero! Are there any plans to adapt this gz-classic feature to gz-garden in the near future? Thanks :)

23pointsNorth, 2023-10-24:
Was wondering about the status of this? Is it a matter of porting it from classic or the underlying changes are large and need a re-write?

azeey, 2023-12-13:
@23pointsNorth, it could be a matter of porting it from classic, but I would think it would be better to add the feature into libsdformat since it's implemented in Gazebo-classic by creating sdformat strings of each model in the population.

scpeters, 2026-09-24:
some folks are ROSCon have expressed interest in this feature as it is useful for education purposes