gazebosim/gz-sim #261: Physics engine tutorial

iche033, 2020-07-25:
Add tutorial showing how to run ign-gazebo with different physics engine

Here's a [related tutorial in ign-physics](https://github.com/ignitionrobotics/ign-physics/blob/master/tutorials/03_physics_plugins.md). We need a high level tutorial for ign-gazebo users showing how to load these physics plugin from cmd line / sdf world file.

chapulina, 2020-07-27:
I think most of this is already covered by this tutorial: https://ignitionrobotics.org/api/gazebo/3.2/physics.html

iche033, 2020-07-27:
> I think most of this is already covered by this tutorial: https://ignitionrobotics.org/api/gazebo/3.2/physics.html
ah yes that's what I was looking for. Closing this issue.

For the people who thumb up this issue. feel free to let us know if there's anything else missing from that tutorial that you like to see

closed 2020-07-27 by iche033