gazebosim/gz-sim #265: Support visual plugins

caguero, 2020-07-28:
Supporting visual plugins that act as GUI overlays is a nice way to customize the GUI to the user needs.

chapulina, 2020-07-28:
Some more pointers:

* `ign-gazebo` should load `<visual><plugin>`s. As a reference, [here](https://github.com/ignitionrobotics/ign-gazebo/compare/fc15b5e5b6d5550d6317b02ded5f1fe6ce001300..6a74c3608c55ba7fbce856478f0836de568e4b51)'s what it took to load `<sensor><plugin>`s.
* Gazebo-classic's visual plugins are loaded by `gazebo::rendering` on both server and client scenes. I believe an equivalent implementation would have to live in `ign-rendering`. If we want this to be in `ign-gazebo`, we may need separate plugins for the server side (`ignition::gazebo::System`) and another one for the client (`ignition::gazebo::GuiSystem`).
* Modifying visuals on the client should already be possible using an `ignition::gazebo::GuiSystem`. It's possible to access the rendering engine's singleton and use `ign-rendering`'s API to modify visuals in the rendering thread, using the `Render` event.
* For accessing the rendering API from a backend system, see https://github.com/ignitionrobotics/ign-gazebo/issues/36

chapulina, 2021-02-03:
The tutorial on #596 explains how to use the rendering API from either server or client.

I believe the only outstanding issue is loading plugins from the `<visual><plugin>` tag.

scpeters, 2021-11-02:
>     * Gazebo-classic's visual plugins are loaded by `gazebo::rendering` on both server and client scenes. I believe an equivalent implementation would have to live in `ign-rendering`.

I've opened https://github.com/ignitionrobotics/ign-rendering/issues/482 to track the ign-rendering side of this.

closed 2022-03-01 by chapulina