gazebosim/gz-sim #273: Touch plugin enabled

mohamedsayed18, 2020-08-05:
The `<enable>` tag is not enabled. I tried to set it to `false` but it always enabled.
https://github.com/ignitionrobotics/ign-gazebo/blob/deaa059e7a82e177279343de125c39b44f09237a/src/systems/touch_plugin/TouchPlugin.cc#L187-L192

This condition is always achieved.
I think the problem is with `Get<bool>`

azeey, 2020-08-05:
Oof, yeah that should be `_sdf->Get<bool>("enabled", false).first`. Mind creating a PR?

mohamedsayed18, 2020-08-05:
I tried to edit the `workspace/src/sdformat/include/sdf/Element.hh` but it didn't work. Is this the write file?

azeey, 2020-08-05:
I meant line 188 https://github.com/ignitionrobotics/ign-gazebo/blob/deaa059e7a82e177279343de125c39b44f09237a/src/systems/touch_plugin/TouchPlugin.cc#L188. `second` should be `first`.

`Get` returns a `std::pair` where the first value is the actual value of the sdf parameter and the second value indicates whether the parameter was set.

PR #275 (2020-08-07): Touch plugin enabled

fix #273
closed 2020-08-07 by chapulina