gazebosim/gz-sim #280: Unable to find URI (Fuel models)

mcres, 2020-08-10:
When I try to include a model from Fuel, for instance this one:
```
<include>
  <pose>0 0 0.7 0 0 0</pose>
  <uri>
    https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/Coke
  </uri>
</include>
```

I get the following error when I `ign gazebo -v 4 myWorld.sdf`, after which the application ends:
```
[Err] [Server.cc:146] Error Code 12 Msg: Unable to find uri[
        https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/Coke
      ]
[Err] [Server.cc:146] Error Code 11 Msg: Invalid uri[
        https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/Coke
      ]. Should be model://
        https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/Coke
```
It's the same error with most of the Fuel models.
However, if I specify the model with meshes, it will load with no problems:
```
<model name="coke_can">
  <pose>0 0 0.7 0 0 0</pose>
  <static>true</static>
  <link name="link">
    <collision name="collision">
      <geometry>
        <mesh>
          <uri>https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/Coke/2/files/meshes/coke.obj</uri>
        </mesh>
      </geometry>
    </collision>
    <visual name="visual">
      <geometry>
        <mesh>
          <uri>https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/Coke/2/files/meshes/coke.obj</uri>
        </mesh>
      </geometry>
    </visual>
  </link>
</model>
```
I didn't have that problem before updating Dome #262 

___________________________________________________________________________________________
For a more detailed information, this is the `sdf` file I'm trying to load:
```
<?xml version="1.0" ?>
<!--
  Demo of models being downloaded from http://app.ignitionrobotics.org/
-->
<sdf version="1.6">
  <world name="shapes">
    <plugin
      filename="libignition-gazebo-physics-system.so"
      name="ignition::gazebo::systems::Physics">
    </plugin>
    <plugin
      filename="libignition-gazebo-scene-broadcaster-system.so"
      name="ignition::gazebo::systems::SceneBroadcaster">
    </plugin>
    <gui fullscreen="0">
      <!-- 3D scene -->
      <plugin filename="GzScene3D" name="3D View">
        <ignition-gui>
          <title>3D View</title>
          <property type="bool" key="showTitleBar">false</property>
          <property type="string" key="state">docked</property>
        </ignition-gui>

        <engine>ogre</engine>
        <scene>scene</scene>
        <ambient_light>0.4 0.4 0.4</ambient_light>
        <background_color>0.8 0.8 0.8</background_color>
        <camera_pose>-6 0 6 0 0.5 0</camera_pose>
      </plugin>

      <!-- World control -->
      <plugin filename="WorldControl" name="World control">
        <ignition-gui>
          <title>World control</title>
          <property type="bool" key="showTitleBar">false</property>
          <property type="bool" key="resizable">false</property>
          <property type="double" key="height">72</property>
          <property type="double" key="width">121</property>
          <property type="double" key="z">1</property>

          <property type="string" key="state">floating</property>
          <anchors target="3D View">
            <line own="left" target="left"/>
            <line own="bottom" target="bottom"/>
          </anchors>
        </ignition-gui>

        <play_pause>true</play_pause>
        <step>true</step>
        <start_paused>true</start_paused>
        <service>/world/shapes/control</service>
        <stats_topic>/world/shapes/stats</stats_topic>

      </plugin>

      <!-- World statistics -->
      <plugin filename="WorldStats" name="World stats">
        <ignition-gui>
          <title>World stats</title>
          <property type="bool" key="showTitleBar">false</property>
          <property type="bool" key="resizable">false</property>
          <property type="double" key="height">110</property>
          <property type="double" key="width">290</property>
          <property type="double" key="z">1</property>

          <property type="string" key="state">floating</property>
          <anchors target="3D View">
            <line own="right" target="right"/>
            <line own="bottom" target="bottom"/>
          </anchors>
        </ignition-gui>

        <sim_time>true</sim_time>
        <real_time>true</real_time>
        <real_time_factor>true</real_time_factor>
        <iterations>true</iterations>
        <topic>/world/shapes/stats</topic>

      </plugin>

    </gui>

    <light type="directional" name="sun">
      <cast_shadows>true</cast_shadows>
      <pose>0 0 10 0 0 0</pose>
      <diffuse>0.8 0.8 0.8 1</diffuse>
      <specular>0.2 0.2 0.2 1</specular>
      <attenuation>
        <range>1000</range>
        <constant>0.9</constant>
        <linear>0.01</linear>
        <quadratic>0.001</quadratic>
      </attenuation>
      <direction>-0.5 0.1 -0.9</direction>
    </light>

    <model name="ground_plane">
      <static>true</static>
      <link name="link">
        <collision name="collision">
          <geometry>
            <plane>
              <normal>0 0 1</normal>
            </plane>
          </geometry>
        </collision>
        <visual name="visual">
          <geometry>
            <plane>
              <normal>0 0 1</normal>
              <size>100 100</size>
            </plane>
          </geometry>
          <material>
            <ambient>0.8 0.8 0.8 1</ambient>
            <diffuse>0.8 0.8 0.8 1</diffuse>
            <specular>0.8 0.8 0.8 1</specular>
          </material>
        </visual>
      </link>
    </model>

    <include>
      <pose>0 0 0.7 0 0 0</pose>
      <uri>
        https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/Coke
      </uri>
    </include>

  </world>
</sdf>
```
And this is the output:
```
[Msg] Ignition Gazebo Server v4.0.0~pre1
[Msg] Ignition Gazebo GUI    v4.0.0~pre1
[Msg] Loading SDF world file[/home/mcres/ignition-dome/src/ign-gazebo/examples/worlds/fuel.sdf].
[Err] [Server.cc:146] Error Code 12 Msg: Unable to find uri[
        https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/Coke
      ]
[Err] [Server.cc:146] Error Code 11 Msg: Invalid uri[
        https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/Coke
      ]. Should be model://
        https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/Coke
      
[Dbg] [ign.cc:269] Shutting down ign-gazebo-server
[Dbg] [Application.cc:87] Initializing application.
[GUI] [Dbg] [Application.cc:407] Create main window
[GUI] [Dbg] [Gui.cc:129] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [SignalHandler.cc:141] Received signal[2].
[GUI] [Dbg] [SignalHandler.cc:141] Received signal[2].
[GUI] [Dbg] [Application.cc:130] Terminating application.

```

nkoenig, 2020-08-17:
I did a fresh install of Dome on Ubuntu 18.04, and ran the following world file:

```
<?xml version="1.0" ?>
<sdf version="1.6">
  <world name="shapes">
    <plugin
      filename="libignition-gazebo-physics-system.so"
      name="ignition::gazebo::systems::Physics">
    </plugin>
    <plugin
      filename="libignition-gazebo-user-commands-system.so"
      name="ignition::gazebo::systems::UserCommands">
    </plugin>

    <plugin
      filename="libignition-gazebo-scene-broadcaster-system.so"
      name="ignition::gazebo::systems::SceneBroadcaster">
    </plugin>
    <gui fullscreen="0">
      <!-- 3D scene -->
      <plugin filename="GzScene3D" name="3D View">
        <ignition-gui>
          <title>3D View</title>
          <property type="bool" key="showTitleBar">false</property>
          <property type="string" key="state">docked</property>
        </ignition-gui>

        <engine>ogre2</engine>
        <scene>scene</scene>
        <ambient_light>1.0 1.0 1.0 1.0</ambient_light>
        <background_color>1 1 1 1</background_color>
        <camera_pose>-6 0 6 0 0.5 0</camera_pose>
      </plugin>

      <!-- World control -->
      <plugin filename="WorldControl" name="World control">
        <ignition-gui>
          <title>World control</title>
          <property type="bool" key="showTitleBar">false</property>
          <property type="bool" key="resizable">false</property>
          <property type="double" key="height">72</property>
          <property type="double" key="width">121</property>
          <property type="double" key="z">1</property>

          <property type="string" key="state">floating</property>
          <anchors target="3D View">
            <line own="left" target="left"/>
            <line own="bottom" target="bottom"/>
          </anchors>
        </ignition-gui>

        <play_pause>true</play_pause>
        <step>true</step>
        <start_paused>true</start_paused>
        <service>/world/shapes/control</service>
        <stats_topic>/world/shapes/stats</stats_topic>

      </plugin>

      <!-- World statistics -->
      <plugin filename="WorldStats" name="World stats">
        <ignition-gui>
          <title>World stats</title>
          <property type="bool" key="showTitleBar">false</property>
          <property type="bool" key="resizable">false</property>
          <property type="double" key="height">110</property>
          <property type="double" key="width">290</property>
          <property type="double" key="z">1</property>

          <property type="string" key="state">floating</property>
          <anchors target="3D View">
            <line own="right" target="right"/>
            <line own="bottom" target="bottom"/>
          </anchors>
        </ignition-gui>

        <sim_time>true</sim_time>
        <real_time>true</real_time>
        <real_time_factor>true</real_time_factor>
        <iterations>true</iterations>
        <topic>/world/shapes/stats</topic>
      </plugin>
    </gui>

    <light type="directional" name="sun">
      <cast_shadows>true</cast_shadows>
      <pose>0 0 10 0 0 0</pose>
      <diffuse>1.0 1.0 1.0 1</diffuse>
      <specular>0.2 0.2 0.2 1</specular>
      <attenuation>
        <range>1000</range>
        <constant>0.1</constant>
        <linear>0.001</linear>
        <quadratic>0.0001</quadratic>
      </attenuation>
      <direction>-0.5 0.1 -0.9</direction>
    </light>

    <include>
      <uri>https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/Coke</uri>
    </include>
  </world>
</sdf>
```

The above downloaded the model and ran successfully. Can you try the above SDF file?

mcres, 2020-08-17:
It loads the Fuel model correctly.
Hmm, could it be the `<uri>` tags?

azeey, 2020-08-18:
@ahcorde pointed out that this is because of the newline in 
```xml
<uri>
   https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/Coke
</uri>
```
as opposed to
```xml
<uri>https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/Coke</uri>
```
This might be related to https://github.com/osrf/sdformat/issues/322

mcres, 2020-08-19:
> This might be related to osrf/sdformat#322

I didn't have this problem with `sdf9` but I do with `sdf10`, so it does seems related.

Is there anything else I can do?