gazebosim/gz-sim #285: Unable to create GUI marker with parent

pohzhiee, 2020-08-12:
# Reproduction
The following was ran on Ubuntu 20.04 using source build, with ign-transport reverted by 1 commit.

In terminal 1:
```bash
ign gazebo shapes.sdf -v
```
In terminal 2:
```bash
ign service -s /marker --reqtype ignition.msgs.Marker --reptype ignition.msgs.Empty --timeout 1000 --req 'action: ADD_MODIFY, type: SPHERE, id: 4, scale: {x:2.0, y:2.0, z:2.0}, parent: "cylinder"'
```
This will result in the following shown in terminal 1:
```
...
[Wrn] [Component.hh:144] Trying to serialize component with data type [N3sdf2v95WorldE], which doesn't have `operator<<`. Component will not be serialized.
[Wrn] [Component.hh:144] Trying to serialize component with data type [N3sdf2v95ModelE], which doesn't have `operator<<`. Component will not be serialized.
[GUI] [Wrn] [Application.cc:649] [QT] file::/ComponentInspector/Vector3d.qml:93:9: QML Image: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Wrn] [Application.cc:649] [QT] file::/ComponentInspector/Vector3d.qml:93:9: QML Image: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Msg] Loading plugin [ignition-rendering3-ogre2]
[GUI] [Wrn] [Application.cc:649] [QT] Qt has caught an exception thrown from an event handler. Throwing
exceptions from an event handler is not supported in Qt.
You must not let any exception whatsoever propagate through Qt code.
If that is not possible, in Qt 5 you must at least reimplement
QCoreApplication::notify() and catch all exceptions there.

terminate called after throwing an instance of 'Ogre::InvalidParametersException'
  what():  OGRE EXCEPTION(2:InvalidParametersException): Node ID: 21, named '' already was a child of Node ID: 8, named ''. in Node::addChild at /var/lib/jenkins/workspace/ogre-2.1-debbuilder/repo/OgreMain/src/OgreNode.cpp (line 440)
```

# Cause
In the [marker creation section of the code](https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo3/src/rendering/MarkerManager.cc#L480-L505),

`this->SetVisual(_msg, visualPtr);` will add the marker to a parent if specified, in this [section of the code](https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo3/src/rendering/MarkerManager.cc#L276-L283).
Then, a few lines later, `scene->RootVisual()->AddChild(visualPtr);`(L501) attempts to add the root visual as the parent of the marker. Since a  cannot have 2 parents, the current issue is observed.

https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo3/src/rendering/MarkerManager.cc#L492

pohzhiee, 2020-08-12:
# Proposed solution
Replace line 501 with the following 3 lines
```c++
      if(!visualPtr->HasParent()){
          this->scene->RootVisual()->AddChild(visualPtr);
      }
```

chapulina, 2020-08-12:
Hi @pohzhiee , thank you for ticketing the issue and the proposed fix. You said you're using a source build, but what branch are you on?

Would you be up for opening a PR with your proposed fix? It would be great if you added a test case that reproduces the issue and is fixed by your patch.

pohzhiee, 2020-08-13:
@chapulina I am using the branches specified in the [citadel yaml](https://raw.githubusercontent.com/ignition-tooling/gazebodistro/master/collection-citadel.yaml), updated to their latest commit as of the time of issue, except ign-transport which was reverted by 1 commit.

I'll try to open a PR to fix this issue. However, I am not aware that there are tests for the GUI right now?

chapulina, 2020-08-13:
> I'll try to open a PR to fix this issue.

Thanks!

> I am not aware that there are tests for the GUI right now?

Oh you're right, it looks like the `MarkerManager` doesn't have any tests right now :confused: We're starting to add tests for GUI plugins on `ign-gui`, but we don't have any yet in `ign-gazebo`.

JShep1, 2020-08-13:
> # Side note
> For some reason the service call will always timeout even though it is successful, is it due to an issue in ign-transport?

I'm going to take a look into this and see what I can find

JShep1, 2020-08-13:
In the meantime while we investigate this issue, you could try looking at the example here https://github.com/ignitionrobotics/ign-gazebo/tree/ign-gazebo3/examples/standalone/marker to be able to generate markers in code.

JShep1, 2020-08-13:
We submitted a fix for the issue you're witnessing here https://github.com/ignitionrobotics/ign-gazebo/pull/292 .  Essentially what was occurring was that no material was being set so the default was previously invisible, the previously mentioned PR sets the default material for a marker visual to white.  You can also specify the color so the marker shows up with something like `ign service -s /marker --reqtype ignition.msgs.Marker --reptype ignition.msgs.Empty --timeout 1000 --req 'action: ADD_MODIFY, type: SPHERE, id: 4, scale: {x:2.0, y:2.0, z:2.0}, parent: "cylinder", material: {ambient: {r:0.0, g:0.0, b:1.0, a:1.0}, diffuse: {r:0.0, g:0.0, b:1.0, a:1.0}}`

The timeout service call message is misleading and can be ignored for now.

pohzhiee, 2020-08-14:
> The timeout service call message is misleading and can be ignored for now.

Although it can be ignored for this issue specifically, it can prove to be problematic when calling this service from code, as there is no way to verify that the service actually succeeded or not. But that is a separate issue with ign-transport and I think i'll try to open an issue there once I can narrow down the problem. 
Anyway, here is my code snippet that is problematic.
```c++
bool result;
ignition::msgs::Empty dummy;
auto requestResult = node->Request("/marker", markerMsg, 3000, dummy, result);
```
This will always have both `request` and `requestResult` being false, even though I can visibly see the marker being spawned.

JShep1, 2020-08-14:
> This will always have both `request` and `requestResult` being false, even though I can visibly see the marker being spawned.

Yes, we've spoken internally about the fix for this issue and it will involve changing the return type of `OnMarkerMsg` ([here](https://github.com/ignitionrobotics/ign-gazebo/blob/master/src/rendering/MarkerManager.cc#L65)) from `void` to `bool`, since this would break ABI, it's a change that will likely occur in Dome.

pohzhiee, 2020-08-14:
> Yes, we've spoken internally about the fix for this issue and it will involve changing the return type of `OnMarkerMsg` [...]

Ok thanks that would be great. Can we close this issue now?

JShep1, 2020-08-14:
> Ok thanks that would be great. Can we close this issue now?

Yep!  Thanks for bringing these issues up, definitely feel free to continue ticketing issues (and fixes if you're feeling generous :wink: ) going forward, we appreciate the help.

closed 2020-08-14 by JShep1