gazebosim/gz-sim #286: PBR file paths cannot be relative

nkoenig, 2020-08-12:
Material maps within an SDF file require `model://`. For example:

```
          <pbr>
            <metal>
              <albedo_map>model://UnderwaterWorld/materials/textures/Rock_Albedo.png</albedo_map>
              <normal_map>model://UnderwaterWorld/materials/textures/Rock_Normal.png</normal_map>
              <roughness_map>model://UnderwaterWorld/materials/textures/Rock_Roughness.png</roughness_map>
            </metal>
          </pbr>
```

It would be nice if these paths could be relative like `<mesh><uri>`.

iche033, 2020-08-12:
This is the place in code that parses these sdf params:
https://github.com/ignitionrobotics/ign-gazebo/blob/master/src/rendering/SceneManager.cc#L428

looks like `common::findFile` is not sufficient. I noticed that other places use `asFullPath`, e.g. for parsing mesh path, so may be that'll help

nkoenig, 2020-09-02:
See #328.

chapulina, 2020-10-19:
Reopening this issue because the fix only affected the client, but not the server. To reproduce the remaining issue, use a world that has rendering sensors, for example:

1. `ign gazebo -v 1 -r sensors_demo.sdf`
1. Drag this model into simulation: https://app.ignitionrobotics.org/chapulina/fuel/models/Rescue%20Randy%20-%20relative%20paths
1. The model will look good on the GUI but not for the sensors. And you'll also see these errors:

```[Err] [SystemPaths.cc:453] Could not resolve file [materials/textures/rescue_randy_roughness.png]
[Err] [SceneManager.cc:450] Unable to find file [materials/textures/rescue_randy_roughness.png]
[Err] [SystemPaths.cc:453] Could not resolve file [materials/textures/rescue_randy_metalness.png]
[Err] [SceneManager.cc:461] Unable to find file [materials/textures/rescue_randy_metalness.png]
[Err] [SystemPaths.cc:453] Could not resolve file [materials/textures/rescue_randy_albedo.png]
[Err] [SceneManager.cc:483] Unable to find file [materials/textures/rescue_randy_albedo.png]
[Err] [SystemPaths.cc:453] Could not resolve file [materials/textures/rescue_randy_normal.png]
[Err] [SceneManager.cc:494] Unable to find file [materials/textures/rescue_randy_normal.png]
```

[image]

chapulina, 2021-11-15:
Sounds like it's fixed

PR #328 (2020-09-16): Support relative paths for PBR materials
Resolves #286.

Requires: https://github.com/osrf/sdformat/pull/349.

To test, you can modify PBR map URIs similar to the following.

Old
```
<pbr>
  <metal>
    <albedo_map>https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/Cave Straight Type A/3/files/materials/textures/Gravel_Albedo.png</albedo_map>
    <normal_map>https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/Cave Straight Type A/3/files/materials/textures/Gravel_Normal.png</normal_map>
    <roughness_map>https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/Cave Straight Type [... cut]
closed 2021-11-15 by chapulina