gazebosim/gz-sim #287: Add service for rendering multiple markers

mcres, 2020-08-12:
Currently, Ignition Gazebo doesn't have a service that supports rendering multiple markers at once.
While working on #229, we've seen that if several markers are needed at a specific moment, requesting them individually could drop the performance significantly due to transport messages costs.

This could be done using the already existing [Marker_V message](https://github.com/ignitionrobotics/ign-msgs/blob/master/proto/ignition/msgs/marker_v.proto) as the request and most of the existing code in `MarkerManager.cc`, resulting in a [service callback](https://github.com/ignitionrobotics/ign-gazebo/blob/master/src/rendering/MarkerManager.cc#L593-L597) very similar to the `/marker` service, which renders an individual marker.

JShep1, 2020-08-20:
Thanks for suggesting this, I created a PR [here](https://github.com/ignitionrobotics/ign-gazebo/pull/302/files) adding this functionality.

mcres, 2020-08-21:
Awesome, thanks for the PR!

chapulina, 2020-08-27:
#302

PR #302 (2020-08-27): Add marker array service
Closes #287
closed 2020-08-27 by chapulina