gazebosim/gz-sim #293: Log Playback example

JShep1, 2020-08-14:
Currently, the log playback example ([here](https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo2/examples/worlds/log_playback.sdf)) obtains a log from `/tmp/log` which the other `log_record*.sdf` no longer store to, it seems the behavior has become deprecated as indicated [here](https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo2/examples/worlds/log_record_dbl_pendulum.sdf#L34-L36).

I feel like we should either update the documentation of the `log_playback.sdf` file or modify it in some way so that going forward, confusion is minimized for the user.  I'm not sure the best way given the logs from the record examples are now stored in a timestamped location so we can't specify a static location for these log files.  Do you have any thoughts @mabelzhang ?

mabelzhang, 2020-08-17:
Maybe the way that makes the most sense is to just comment it out with a note saying this tag is valid and can be used.

I originally thought that tag isn't being used again but I checked the code and apparently it is. Looks like the tutorial documents both the `--playback <path>` option and the SDF option, so it should be okay: https://github.com/ignitionrobotics/ign-gazebo/blob/master/tutorials/log.md

mabelzhang, 2020-08-19:
I looked again, and I think we should just remove `log_playback.sdf`.

The reason is that `<path>` is actually injected by `Server.cc` here https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo2/src/Server.cc#L87-L90
Its value comes from `_config.LogPlaybackPath()`, which is the value specified on the command line with `--playback`. So there's really no need for the user to create an SDF just for playback.

Specifying the path like in `log_playback.sdf` still works, but to be parallel with recording, which deprecated `<path>`, I wonder if we should also deprecate it in the playback. In that case, we should remove the playback SDF example in the log tutorial too.

In the state the file is right now, if we just comment out `<path>`, running the file will result in an error:
```
[Err] [LogPlayback.cc:247] Unspecified log path to playback. Nothing to play.
```
If we leave in the `<path>`, running the file also results in an error:
```
[Err] [LogPlayback.cc:256] Log path invalid. File [/tmp/log/state.tlog] does not exist. Nothing to play.
```
So I don't see a reasonable way to keep the file. The only thing in that file is really those few lines that `Server.cc` populates.

JShep1, 2020-08-20:
Thanks for taking a look, I agree with your thoughts.  Do you believe these changes, most particularly deprecating `<path>`, would affect SubT in anyway, @nkoenig ?

mabelzhang, 2020-10-22:
Dome progress for the record:

I made the code change for Dome, but now I have something really weird going on with the `LogControl` test.

Something seems to be propagating down from the `RecordAndPlayback` test right before it, in that if I comment out the `RecordAndPlayback` test, then `LogControl` passes. Otherwise `LogControl` fails, trying to load a path `test/test_logs/test_logs_playback/state.tlog`, which the test isn't even supposed to have knowledge about! I don't know where it catches wind of that file, but it's supposed to be loading a completely different one.

I've poured a few hours into this today. It seems crazy. I don't know if I'll have time to look at it again until next week.

mabelzhang, 2020-12-04:
I've found the root cause of the weird behavior that a variable from an earlier test propagates to the next test.

It has to do with `DefaultWorld::World::world` being `static` in Server.cc:
https://github.com/ignitionrobotics/ign-gazebo/blob/a5abacfa74db6e8db0218bd32c55980e15cf249b/src/Server.cc#L105-L110

Printouts showed me that `DefaultPlugins(_config)` is only called in the earlier test. In the later test, the static function `World` is called, but the static function `DefaultPlugins` is not called, because the static variable `world` has already been initialized.

The result is that the second test inherits the value of `world` from the first test. In this case, that means `<playback_path>` (or simply `<path>` in the current code). That's why the second test is trying to load the playback path from the first test, doesn't find it, and fails.

@chapulina Is the static variable `world` by design? It seems to imply that, we allow only one default world to be running, however many Server instances we create. It also means if a later server instance has a different `ServerConfig`, which is passed to `DefaultPlugins`, then it will not take any effect. The good thing is, that config is only used for injecting the playback path, so maybe there's a different way to specify this path instead of the default SDF?

Just double-checking that it isn't an error, whether we need to change anything in `Server.cc`. If that's by design, I'll find a way to work around it in the tests.

chapulina, 2020-12-04:
Wow, nice catch! I think the `World` function is meant to be static, but the `world` variable inside it being static looks like a mistake to me.

mabelzhang, 2021-01-16:
Now that #537 is in, the above `static` problem disappeared (the code, literally) :)

mabelzhang, 2021-03-03:
Closing issue as fixes have been merged in Blueprint, Citadel, and Dome.

closed 2021-03-03 by mabelzhang