gazebosim/gz-sim #294: ign topic -l empty in ignition-blueprint

peci1, 2020-08-14:
We have one desktop computer with clean install of Ubuntu 18.04.4 and NVidia GTX 1060 GPU. On this particular desktop, ignition topics of gazebo do not work. I've no idea why, because on other machines it does.

Terminal 1:

    $ ign gazebo

Terminal 2:

    $ ign topic -l
    $

Clearly, the `ign topic` command doesn't see any published topics. On other machines it does.

But if I try ign-transport directly, it works:

Terminal 1:

    $ while true; do ign topic -t /foo -m   ignition.msgs.StringMsg -p 'data:"Custom data"'; done

Terminal 2:

    $ ign topic -e -t /foo
    data: "Custom data"

    data: "Custom data"

    data: "Custom data"

    data: "Custom data"

    ^C$

Do you have an idea how to debug this computer? I've tried running in both terminals with `IGN_PARTITION=foo`, but the result was the same. I trued launching with `IGN_VERBOSE=1`, but nothing weird would be shown. I tried calling `ign topic -e` via strace, and I saw a lot of successful `setsockopt`, `IP_ADD_MEMBERSHIP` etc, followed by writing `\1\0\0\0\0\0\0\0\0` into the socket, and then a timed-out poll. I tried turning off all network interfaces except `lo`, but no change. We tried reinstalling the computer - no help.

We've also accidentally installed ignition-citadel, and there the communication worked. But we need blueprint for the SubT challenge. Once I uninstalled citadel (`apt remove ignition-citadel && apt autoremove`), the communication in blueprint is broken again.

caguero, 2020-08-14:
I suspect that your `ign` commands are using a different version of Ignition Transport than Ignition Gazebo. You can check with `ign topic --versions`. If that's the case, you could:

a) Force your `ign topic` to use the same version as Ignition Gazebo. E.g.: `ign topic -l 00force-version 7.5.0`.
b) Remove the leftover files from your previous Ignition Citadel installation.  E.g.: `/usr/share/ignition/transport8.yaml`

peci1, 2020-08-14:
Wow, it actually worked, thank you! So it seems `ignition-citadel` did not uninstall itself correctly. But I tried to replicate it on another computer (install and uninstall citadel) and the leftover file was not left there.

caguero, 2020-08-14:
I'll close the issue but feel free to reopen if needed.

closed 2020-08-14 by caguero

peci1, 2021-02-02:
Another reason why this can happen is if the publishers are run with `IGN_TRANSPORT_TOPIC_STATISTICS=1` and the list command is run with this variable set to zero (or the other way round). As is the case after https://github.com/osrf/subt/pull/742 in SubT simulator. This took me about an hour to figure out :( Isn't there really a way to provide better user experience, at least in case the result of `ign topic -l` is empty?

peci1, 2021-02-02:
I logged an improvement request in ignitionrobotics/ign-transport#218 .