gazebosim/gz-sim #307: Support save a world to collada

nkoenig, 2020-08-24:
It would be nice to support exporting a whole world to a single collada file. This would let us share world configurations with people who don't have Ignition, and we could also (easily?) load worlds into 3D viewers/editors.

There is some support here: https://github.com/ignitionrobotics/ign-common/blob/master/graphics/include/ignition/common/ColladaExporter.hh

nkoenig, 2020-09-24:
@gonzodepedro , we have someone interested in saving the world to collada even if materials are not supported. I'm not sure how close you are to completing material support. You could make an initial PR(s) with just mesh support sans materials.

gonzodepedro, 2020-09-24:
Sure. I'm working on a PR for a plugin and some changes that may be required in [ColladaExporter](https://github.com/ignitionrobotics/ign-common/blob/master/graphics/include/ignition/common/ColladaExporter.hh), I'll link those here when ready, so we can start discussing implementation details.

gonzodepedro, 2020-09-28:
[Here](https://github.com/gonzodepedro/world_exporter) is the code to the plugin.
Requires PR ignitionrobotics/ign-gazebo#307

chapulina, 2020-09-28:
Thanks, @gonzodepedro ! Do you think you could open a PR against this repository adding the plugin to `src/systems`? It would be nice to make it available to anyone installing `ign-gazebo`

chapulina, 2021-02-26:
Done on #474

closed 2021-02-26 by chapulina