gazebosim/gz-sim #310: Grid doesn't update until mouse hovers over 3d window.

nkoenig, 2020-08-25:
To replicate:

1. `ign gazebo -v 4`
2. Open the `Grid Config` plugin.
3. Change any property.
4. The viewport doesn't update with the change until the mouse enters.

chapulina, 2020-08-25:
I think that the actual behaviour is that the grid is not updated until the editing is "finished" on the field. For example, try editing a field and then clicking on another field, or editing a field and pressing enter.

We could change the behaviour to update at every value change instead of editing finish.

claireyywang, 2020-09-09:
To emit signal when value is changed, we need to use `qt.control 2.15` or `2.2` which will change the design of Spinbox from 
[image] to 
[image]

[`onValueModified` signal](https://doc.qt.io/qt-5.9/qml-qtquick-controls2-spinbox.html#valueModified-signal) is only available in `qt.controls. 2.15` onward, and before that the only signal available to spinbox is [`onEditFinished` ](https://doc.qt.io/qt-5/qml-qtquick-controls-spinbox.html#editingFinished-signal)which is what we are using now. If we want to use the latest signal feature which would resolve this issue, then a change in spinbox style is needed. (comparison shown below)

[image]

@nkoenig  @chapulina what do you think?

chapulina, 2020-09-10:
I think QtControls 2's `SpinBox` doesn't really look suitable for desktop interfaces. I think that the `-` and `+` next to each single spin box on that widget will just look confusing.

I'm in favor of just keeping the current behaviour, which I don't think is too uncommon? What do you think, @nkoenig ?

I'd also like to point out that all our widgets are using the same spin box and thus have the same behaviour. If we want the interface to feel consistent we'd need to change them all.

chapulina, 2021-05-04:
I'm closing this because the recommended usage is to press enter to see the changes.

closed 2021-05-04 by chapulina