gazebosim/gz-sim #312: Grid3D crashes

nkoenig, 2020-08-25:
1. `ign gazebo -v 4`
2. Insert the `Grid 3D` plugin.
3. The screen changes to this:

[image]

4. Clicking on the window causes it to crash

Also, what is the difference between `Grid Config` and `Grid 3d`?

claireyywang, 2020-08-25:
I think grid 3d is the outdated grid plugin that is supposed to be replaced by grid config. `grid_3d` is `ign-gui` plugin. `grid_config` is `ign-gazebo` plugin. I remember @chapulina and I talked about this a while ago. Maybe we can remove `grid_3d` to avoid confusion?

chapulina, 2020-08-25:
Yeah `ign-gui`'s `Grid3D` works with `ign-gui`'s `Scene3D`. I think we should move `ign-gazebo`'s `GridConfig` to `ign-gui` and make it work with both `ign-gui`'s scene and `ign-gazebo`'s scene.

chapulina, 2021-02-02:
This should be fixed by #598

chapulina, 2021-12-13:
I think https://github.com/ignitionrobotics/ign-gui/pull/324 completely closes this.

closed 2021-12-13 by chapulina