gazebosim/gz-sim #313: Model info CLI

nkoenig, 2020-08-25:
It would be nice to have command line tool that dumps information about a model. For example:

```
ign model -i <MODEL_NAME>
```
Could return:
1. Bounding box dimensions.
2. Inertia
3. Author, license, version (if applicable)

diegoferigo, 2020-08-25:
This is a great idea, some other nice-to-have info:

- Links
- Joints (plus their dofs and parent / child links)
- Additional frames (thanks to the recent frame semantics)
- Canonical link

chapulina, 2020-08-25:
As a reference, here's Gazebo-classic's tool:

```
$ gz model -h
 gz model [options]

	Change properties of a model, delete a model, or
	spawn a new model. If a name for the world, option -w, is
	not pecified, the first world found on the Gazebo master
	will be used.

Options:
  --verbose               Print extra information
  -h [ --help ]           Print this help message
  -m [ --model-name ] arg Model name.
  -w [ --world-name ] arg World name.
  -d [ --delete ]         Delete a model.
  -f [ --spawn-file ] arg Spawn model from SDF file.
  -s [ --spawn-string ]   Spawn model from SDF string, pass by a pipe.
  -i [ --info ]           Output model state information to the terminal.
  -p [ --pose ]           Output model pose as a space separated 6-tuple: x y z
                          roll pitch yaw.
  -x [ --pose-x ] arg     x value
  -y [ --pose-y ] arg     y value
  -z [ --pose-z ] arg     z value
  -R [ --pose-R ] arg     roll in radians.
  -P [ --pose-P ] arg     pitch in radians.
  -Y [ --pose-Y ] arg     yaw in radians.
```

The spawn / delete / translate functionality is already covered by service calls (`/create`, `/remove`, `/set_pose`).

WagnerMarcos, 2021-06-04:
Hi everyone, I'll be taking this issue now if that's ok. I'll let you know any prrogress

francocipollone, 2021-06-10:
When implementing the `ign model` command, there is some info that could be gotten by using the topics and services that the gazebo server advertises.

However, is there a way to get a reference of the running server or the entity component manager in order to get more information about the models in the simulation?.

WagnerMarcos, 2021-06-11:
I’m currently using the [/world/<used_world>/scene/info] service to get a [Scene proto msg](https://github.com/ignitionrobotics/ign-msgs/blob/ign-msgs7/proto/ignition/msgs/scene.proto#L35) with the current models. Inside of the scene message there is a [Model proto msg](https://github.com/ignitionrobotics/ign-msgs/blob/ign-msgs7/proto/ignition/msgs/model.proto#L35) with some of the desired information. [Here](https://github.com/ignitionrobotics/ign-gazebo/blob/wagnermarcos/CLI_model_info_dump/src/ign.cc#L52) I am trying to get from the Scene proto msg specific fields (at the moment I am dumping the whole message as I couldn’t iterate through the fields yet). Would you have any ideas on how to get a reference to the ECM to request the remaining information about the current models displayed? Or do you know if there is another module that could be useful for this issue?

francocipollone, 2021-06-15:
There is some information that we can get with what is already provided through services and there is some information that needs to be gotten somehow.
During the weekly meeting, some approaches were discussed:
 - Provide another service (or improve the current one) to obtain more information. (affects SceneBroadcaster?)
 - Create a custom system and get it directly from components.
 - Hold a snapshot of the ecm, (not sure how to get it).
 - Some other options may be mentioned.
 
-----------------------------------

I see value in tackling this incrementally:
1. Create an `ign model` to allow get information on demand.
2. Create options to get the information only that is already available to be gotten: e.g: Pose, links, joints.. Probably having a `--list` option to list all the models is helpful.
3. Create options to get information for the rest of what is requested in the issue
  a. This will imply taking one of the aforementioned approaches to "make public" more information about the running simulation.

In order to be more atomic and to keep this in reviewable PRs probably 1 and 2 could be tackled into a single PR and leave 3 for a follow-up PR. Wdyt @chapulina ? In the meantime, I will continue thinking about the better approach to tackle point 3. 

 ------------------------
As updates. 1 and 2 are partially solved in @WagnerMarcos 's [branch](https://github.com/ignitionrobotics/ign-gazebo/tree/wagnermarcos/CLI_model_info_dump) . 
e.g: Tested with loading shapes.sdf -> `ign gazebo shapes.sdf`
```sh
# To get the pose
>> ign model info -m box --pose
position {
  z: 0.5
}
orientation {
  w: 1
}
```
Wondering why the pose is only showing a `z` value though.

chapulina, 2021-06-16:
> Hold a snapshot of the ecm, (not sure how to get it).

See the example added in https://github.com/ignitionrobotics/ign-gazebo/pull/859, which hopefully sheds some light into this approach. It takes advantage of the mechanism that we already have to serialize all simulation state into a single message. The power in this approach is that all information that's available to systems is immediately also available to any external program.

WagnerMarcos, 2021-06-18:
I have been taking the ecm approach, as shown in the example (which was really helpful), getting the state of the simulation. 

At the moment I have been able to get information about the pose and links of a passed model. I have also added the option to list the available models in the simulation. 

Now, when running `diff_drive.sdf` with `ign gazebo diff_drive.sdf`, you will get the following output when running
`>> ign model info -m vehicle_green`

<details>
  <summary>Click to see output</summary>

```
Name: vehicle_green
  - Pose: 
      [0.000000 | -2.000000 | 0.325000]
      [0.000000 | -0.000000 | 0.000000]

  - Link [25]
    - Name: chassis
    - Parent: vehicle_green [24]
    - Mass: [1.143950]
    - Inertial Matrix: 
        [0.126164 | 0.000000 | 0.000000]
        [0.000000 | 0.416519 | 0.000000]
        [0.000000 | 0.000000 | 0.481014]
    - Pose: 
        [-0.151427 | 0.000000 | 0.175000]
        [0.000000 | -0.000000 | 0.000000]
  - Link [28]
    - Name: left_wheel
    - Parent: vehicle_green [24]
    - Mass: [2.000000]
    - Inertial Matrix: 
        [0.145833 | 0.000000 | 0.000000]
        [0.000000 | 0.145833 | 0.000000]
        [0.000000 | 0.000000 | 0.125000]
    - Pose: 
        [0.554283 | 0.625029 | -0.025000]
        [-1.570700 | 0.000000 | 0.000000]
  - Link [31]
    - Name: right_wheel
    - Parent: vehicle_green [24]
    - Mass: [2.000000]
    - Inertial Matrix: 
        [0.145833 | 0.000000 | 0.000000]
        [0.000000 | 0.145833 | 0.000000]
        [0.000000 | 0.000000 | 0.125000]
    - Pose: 
        [0.554282 | -0.625029 | -0.025000]
        [-1.570700 | 0.000000 | 0.000000]
  - Link [34]
    - Name: caster
    - Parent: vehicle_green [24]
    - Mass: [1.000000]
    - Inertial Matrix: 
        [0.100000 | 0.000000 | 0.000000]
        [0.000000 | 0.100000 | 0.000000]
        [0.000000 | 0.000000 | 0.100000]
    - Pose: 
        [-0.957138 | 0.000000 | -0.125000]
        [0.000000 | -0.000000 | 0.000000]
```

</details>

And when running `ign model info --list`
```
Available models:
    - ground_plane
    - vehicle_blue
    - vehicle_green
```

---

For the next steps I will make options select properties, so we can get the whole model obtainable info using
` ign model info -m vehicle_green` (The info option may change)

  - Pose
  - Links
    - Name_link
    - Pose_link
    - Mass
    - Inertia
  - Joints
    - Name
    - Joint type
    - Parent link name
  - BoundingBox
    - min / max corner vector
    - x,y,z length
    - box size
    - center of the box

But it will be possible to get only some of the properties when calling the command as `ign model info -m sphere --pose/links/joints/bounding_box`

Or even a certain link or joint if passing the name as `ign model info -m sphere --link=box_chassis` resulting in an output showing the properties of the passed component.

WagnerMarcos, 2021-06-18:
What I have some doubts about is which properties of each component I should show. For example, for the [links](https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo3/include/ignition/gazebo/Link.hh) I could also get if its a canonical link, if it has self-collide enabled, if windmode is one. 
For the [inertial](https://github.com/ignitionrobotics/ign-math/blob/ign-math6/include/ignition/math/Inertial.hh) component I could get both the mass and moment of inertia matrix, along with the pose vector. Should all of that be printed? 
I am yet to see how to get the canonical links and the additional frames.

Do you think it would be a good idea to make a first PR with a reduced list of these properties?

chapulina, 2021-06-18:
> Do you think it would be a good idea to make a first PR with a reduced list of these properties?

Yeah I think it would be good to make a 1st PR with a minimal set of features to validate the approach, and after that gets in, we can add functionality little by little :+1:

chapulina, 2021-07-29:
The tool was added in #893. I'll consider this issue complete since I understand the main idea was to add the tool, not necessarily to provide specific data. If there's need for specific data to be added, please ticket new issues.

closed 2021-07-29 by chapulina