gazebosim/gz-sim #322: SdfGenerator_TEST failures on master

scpeters, 2020-09-01:
There have been several recurring `UNIT_SdfGenerator_TEST` failures on the master branch since the following build on August 21:

* [image] https://build.osrfoundation.org/job/ignition_gazebo-ci-master-bionic-amd64/24/

The following test failure provides a clue of the difference:

* https://build.osrfoundation.org/job/ignition_gazebo-ci-master-bionic-amd64/24/testReport/junit/(root)/ElementUpdateFixture/WorldWithModelsIncludedExpanded/

The original world file has the following uri:

~~~
<uri>https://fuel.ignitionrobotics.org/1.0/openrobotics/models/Backpack/1/files/meshes/Backpack.dae</uri>`
~~~

The generated world file appears to have 2 URI's concatenated together:

~~~
<uri>file:///root/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/Backpack/1/https://fuel.ignitionrobotics.org/1.0/openrobotics/models/Backpack/1/files/meshes/Backpack.dae</uri>
~~~

I added some debugging print statements to the test in https://github.com/scpeters/ign-gazebo/commit/a1490fbd896024e7083bc3c9629e5e79b7165469 that confirm the other failures are connected to this same issue.

I think this may be related to https://github.com/ignitionrobotics/ign-fuel-tools/pull/85

scpeters, 2020-09-01:
@ahcorde noticed this same thing in https://github.com/ignitionrobotics/ign-gazebo/pull/314#issuecomment-680980828

chapulina, 2020-09-01:
They should be fixed when #271 is merged forward, see #321

chapulina, 2020-09-15:
Latest master branch build is blue: https://build.osrfoundation.org/job/ignition_gazebo-ci-master-bionic-amd64/29/

closed 2020-09-15 by chapulina