gazebosim/gz-sim #324: Error launching ign gazebo without world name

jennuine, 2020-09-01:
Using Ubuntu Focal.

When using citadel and launching `ign gazebo` then clicking one of the shapes (e.g., sphere, box, cylinder), unable to insert it into the world. The output error: `Service [/world//create] is not valid.` This occurs when `ign gazebo` is launched without a world name.

Running `ign gazebo -v 4 shapes.sdf` resolves this issue.

chapulina, 2020-09-01:
How did you install Citadel? Can you check the exact version of `ign-gazebo` and `ign-gui` that you're using? 

Also, could you try deleting this file and running again? `rm ~/.ignition/gazebo/gui.config`

jennuine, 2020-09-01:
I installed Citadel from source following the tutorials found [here](https://ignitionrobotics.org/docs/citadel/install_ubuntu_src). The version for both is 3.2.0

Deleting the `~/.ignition/gazebo/gui.config` file and running it again resolved the issue

iche033, 2020-09-01:
@jennuine did you update your ign-gazebo version between runs? I'm wondering this issue would happen to new users who install ign-gazebo3 the first time.

jennuine, 2020-09-01:
@iche033 do you mean the entire workspace where citadel src is or do you mean the ign-gazebo package? If you mean the ign-gazebo package then no I did not. The only updates to the workspace were I added the ros_ign package and switched to the copy_buffer branch in ign-rendering.

iche033, 2020-09-01:
> @iche033 do you mean the entire workspace where citadel src is or do you mean the ign-gazebo package?

mainly ignition-gazebo and ignition-gui but looks like you haven't updated those two. Initially I thought it's due to recent changes in the code but since you're running the same version every time, there maybe an issue with the initial generated `~/.ignition/gazebo/gui.config`? @chapulina  may know more about this.

chapulina, 2020-09-02:
> I'm wondering this issue would happen to new users who install ign-gazebo3 the first time.

Yeah the 3.3.0 releases should have fixed the situation. The main change is https://github.com/ignitionrobotics/ign-gui/pull/111.

> Deleting the ~/.ignition/gazebo/gui.config file and running it again resolved the issue

Glad to hear. I'll close this issue but please re-open if the issue comes back.

closed 2020-09-02 by chapulina