gazebosim/gz-sim #326: Conan & Ignition

joxoby, 2020-09-02:
Hello everyone,

I have started the project of making the Ignition libraries available as [Conan](https://docs.conan.io/en/latest/) packages. The goal is to alleviate some of the dependencies issues and make development for Ignition easier.

The repository is: https://github.com/joxoby/ign-conan. Feel free to contribute if you're interested. My plan is to get a solid set of Ignition libraries working and then push the resulting recipes to [conan-center-index](https://github.com/conan-io/conan-center-index).

chapulina, 2020-09-02:
Hi @joxoby , thank you for the initiative, that's great to hear! :+1:  I think a post to Gazebo Community would be best to give this effort the visibility it deserves. Then we can close this issue, and you be sure to ticket any new issues if you encounter problems while you progress.

joxoby, 2020-09-02:
New post here: https://community.gazebosim.org/t/packaging-ignition-with-conan/639

closed 2020-09-02 by joxoby