gazebosim/gz-sim #329: Spot light does not use <direction> tag.

nkoenig, 2020-09-02:
According to SDF, a spot light should use the [`<direction>` tag](https://github.com/osrf/sdformat/blob/master/sdf/1.8/light.sdf#L43).

Gazebo doesn't seem to follow this guidance. See [here](https://github.com/ignitionrobotics/ign-gazebo/blob/master/src/rendering/SceneManager.cc#L863).

Not sure if SDF is wrong, or Gazebo.

chapulina, 2020-09-02:
I always thought that the `direction` tag added extra unnecessary complexity. Lights already have a pose, which should be enough to define where they're looking at.

iche033, 2020-09-02:
I remember in gazebo-classic the pose tag applies additional rotation to the direction vector. We can keep the behavior consistent

chapulina, 2021-08-03:
Done in #718