gazebosim/gz-sim #344: No grid is visible

Reka-Berci-Hajnovics, 2020-09-10:
After running `ign gazebo shapes.sdf` there is no grid displayed. I've tried to add the **Grid config plugin** and play around with the parameters, but no grid appears. 
Do you have any suggestions what can be the reason?
I have a fresh installation of foxy and Ignition gazebo version 3.3.0
[image]

chapulina, 2020-09-10:
Do you see any errors or warnings if you run with `-v 2`?

Also, what operating system are you on?

Reka-Berci-Hajnovics, 2020-09-11:
This is the output of ign gazebo shapes.sdf -v 2 
`[Wrn] [URI.cc:580] Unable to parse URI [shapes.sdf]. Ignoring.
[GUI] [Wrn] [Application.cc:649] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:102:33: QML Settings: Failed to initialize QSettings instance. Status code is: 1
[GUI] [Wrn] [Application.cc:649] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:102:33: QML Settings: The following application identifiers have not been set: QVector("organizationName", "organizationDomain")
[GUI] [Wrn] [Application.cc:649] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:102:33: QML Settings: Failed to initialize QSettings instance. Status code is: 1
[GUI] [Wrn] [Application.cc:649] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:102:33: QML Settings: The following application identifiers have not been set: QVector("organizationName", "organizationDomain")
[GUI] [Wrn] [Application.cc:649] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:102:33: QML Settings: Failed to initialize QSettings instance. Status code is: 1
[GUI] [Wrn] [Application.cc:649] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:102:33: QML Settings: The following application identifiers have not been set: QVector("organizationName", "organizationDomain")
[GUI] [Wrn] [Application.cc:649] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:102:33: QML Settings: Failed to initialize QSettings instance. Status code is: 1
[GUI] [Wrn] [Application.cc:649] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:102:33: QML Settings: The following application identifiers have not been set: QVector("organizationName", "organizationDomain")
[GUI] [Wrn] [Application.cc:649] [QT] file::/WorldControl/WorldControl.qml:88:3: QML RoundButton: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Wrn] [Application.cc:649] [QT] file::/WorldControl/WorldControl.qml:88:3: QML RoundButton: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[Wrn] [Component.hh:144] Trying to serialize component with data type [N3sdf2v95WorldE], which doesn't have `operator<<`. Component will not be serialized.
[Wrn] [Component.hh:144] Trying to serialize component with data type [N3sdf2v95ModelE], which doesn't have `operator<<`. Component will not be serialized.
[GUI] [Wrn] [Application.cc:649] [QT] file::/ComponentInspector/Vector3d.qml:93:9: QML Image: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Wrn] [Application.cc:649] [QT] file::/ComponentInspector/Vector3d.qml:93:9: QML Image: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Wrn] [RenderUtil.cc:1073] Failed to create grid for scene [scene] on engine [ogre2]`

I'm running Ubuntu 20.04

chapulina, 2020-09-11:
Oh ok, I see that the grid code for Ogre 2 hasn't been released into the latest `ign-rendering` Citadel release, [3.1.0](https://github.com/ignitionrobotics/ign-rendering/blob/3f163c7ac317bcba6db36aaab18b222dc025be79/ogre2/src/Ogre2Scene.cc#L360-L365). But it's there on the `ign-rendering3` [branch](https://github.com/ignitionrobotics/ign-rendering/blob/640d6236dfc58299fae166965b6a9b721dcd8240/ogre2/src/Ogre2Scene.cc#L361-L367), so it will come on the next minor release.

The grid should work if you use Ogre 1 though, just change `ogre2` to `ogre` in your SDF file.

Reka-Berci-Hajnovics, 2020-09-14:
Great! How can I change `ogre2` to `ogre`? I can't find any link to `ogre ` in the SDF file. I'm running  `ign gazebo shapes.sdf` and I was looking for `ogre` in the /`usr/share/ignition/ignition-gazebo3/worlds/shapes.sdf ` file.

Reka-Berci-Hajnovics, 2020-09-14:
After following the "Build your own robot" tutorial I've found the answer to my question.
I'll post it step-by-step, so it'll be easy to follow if someone else runs into the same problem.
1. I've created a directory for the edited sdf files:
`mkdir ~/citadel/tutorials`
2. I've copied the `shapes.sdf` to this directory:
`cp /usr/share/ignition/ignition-gazebo3/worlds/shapes.sdf ~/citadel/tutorials/
`
3. I've opened the shapes.sdf file and inserted the GUI configuration from the SDF file created in the build your own robot tutorial.

4. Then I've changed ogre2 to ogre.
5. The new file can be run by movign the directory of the file.

Here is the edited shapes.sdf:
`<?xml version="1.0" ?>

<sdf version="1.6">
  <world name="shapes">
    <physics name="1ms" type="ignored">
      <max_step_size>0.001</max_step_size>
      <real_time_factor>1.0</real_time_factor>
    </physics>
    <plugin
      filename="libignition-gazebo-physics-system.so"
      name="ignition::gazebo::systems::Physics">
    </plugin>
    <plugin
      filename="libignition-gazebo-user-commands-system.so"
      name="ignition::gazebo::systems::UserCommands">
    </plugin>
    <plugin
      filename="libignition-gazebo-scene-broadcaster-system.so"
      name="ignition::gazebo::systems::SceneBroadcaster">
    </plugin>
    
    
 <gui fullscreen="0">

            
            <plugin filename="GzScene3D" name="3D View">
                <ignition-gui>
                <title>3D View</title>
                <property type="bool" key="showTitleBar">false</property>
                <property type="string" key="state">docked</property>
                </ignition-gui>

                <engine>ogre</engine> 
                <scene>scene</scene>
                <ambient_light>0.4 0.4 0.4</ambient_light>
                <background_color>0.8 0.8 0.8</background_color>
            </plugin>

            
            <plugin filename="WorldControl" name="World control">
                <ignition-gui>
                <title>World control</title>
                <property type="bool" key="showTitleBar">false</property>
                <property type="bool" key="resizable">false</property>
                <property type="double" key="height">72</property>
                <property type="double" key="width">121</property>
                <property type="double" key="z">1</property>

                <property type="string" key="state">floating</property>
                <anchors target="3D View">
                    <line own="left" target="left"/>
                    <line own="bottom" target="bottom"/>
                </anchors>
                </ignition-gui>

                <play_pause>true</play_pause>
                <step>true</step>
[... 172 lines cut ...]
        </inertial>
        <collision name="sphere_collision">
          <geometry>
            <sphere>
              <radius>0.5</radius>
            </sphere>
          </geometry>
        </collision>

        <visual name="sphere_visual">
          <geometry>
            <sphere>
              <radius>0.5</radius>
            </sphere>
          </geometry>
          <material>
            <ambient>0 0 1 1</ambient>
            <diffuse>0 0 1 1</diffuse>
            <specular>0 0 1 1</specular>
          </material>
        </visual>
      </link>
    </model>
  </world>
</sdf>`

chapulina, 2020-09-14:
> How can I change ogre2 to ogre?

Glad you figured out how to do it from the SDF file. It will soon be possible to do it from the command line, keep an eye on #227

chapulina, 2020-10-16:
I just tested with the latest Citadel and Dome releases and the grid shows:

`ign gazebo --force-version 3.4.0 shapes.sdf`

`ign gazebo --force-version 4.0.0 shapes.sdf`

closed 2020-10-16 by chapulina