gazebosim/gz-sim #345: Translation and rotation doesn't work as expected

Reka-Berci-Hajnovics, 2020-09-10:
In the translation mode the x-axis arrow gets highlighted when I click on the mid of the z-axis arrow. The y-axis and z-axis arrows don't become active after clicking on them.
In the rotation mode a similar thing happens. The roll becomes active after clicking a couple of centimeters below the circle. The other directions do not become active.
I have a fresh installation of ignition gazebo version 3.3.0, foxy, and ubuntu 20.04.
[image]
[image]
[image]
[image]

chapulina, 2020-09-10:
This sounds related to #147. Are you on a high-dpi display, and are you inside a Docker container?

iche033, 2020-09-11:
what graphics driver do you have?

Reka-Berci-Hajnovics, 2020-09-11:
It works now! I'm running Ubuntu 20.04 and a high-dpi display. I've enabled the fractional scaling in the Ubuntu Settings-> Display. I haven't change the scale value, but already by enabling it now I can select the arrow to translate and the circle to rotate the objects in Ignition!

chapulina, 2020-09-11:
Glad to hear it's solved, I'll close this issue, feel free to reopen if the problem comes back.

It may be a good idea to add a note about this to the [troubleshooting documentation](https://ignitionrobotics.org/docs/citadel/install_ubuntu_src#troubleshooting).

closed 2020-09-11 by chapulina

iche033, 2020-09-11:
thanks for posting the fix. 

just curious, what's the scale value that was set?

Reka-Berci-Hajnovics, 2020-09-14:
I haven't change the scale value, it was left on 100%. 
This morning I've started ign gazebo shapes.sdf and the translation and rotation weren't working, neither on my built-in screen nor on my external screen.  I've tried to unplug my external screen and then it was working on my built-in screen. 
After plugging in my external screen it still works now. I've tried to stop and relaunch the application and it is still working. I find it a pity that it is not clear what causes this issue. If I'll find something else what might be related to it I'll keep you posted.

iche033, 2020-09-14:
ok got it, thanks for keeping an eye on it. 

It could be the way we're [querying the scaling factor](https://github.com/ignitionrobotics/ign-rendering/blob/master/src/Utils.cc#L32) in ignition. Let me know if you have more info and I'll see if we can fix this programmatically, thanks.