gazebosim/gz-sim #347: Record entity count in a log file.

nkoenig, 2020-09-11:
We need an entity ID offset, stored in a log file, in order to support spawning of new objects during log playback.

Logplayback will directly insert entities with their recorded ids, which doesn't alter the entityCount. As a result, a spawn call will attempt to create an entity with an existing ID.

A work-around is in #346 [here](https://github.com/ignitionrobotics/ign-gazebo/pull/346/files#diff-6b03084fc9b602934ef716fda06fc9a9R214).

chapulina, 2020-09-11:
> support spawning of new objects during log playback.

What's the use case? Is this for visualization purposes? Typically, during log playback, physics and other systems are not running.

nkoenig, 2020-09-11:
The initial use case is to add lights during log playback.

nkoenig, 2020-09-11:
We will likely also want to spawn models with camera sensors.