gazebosim/gz-sim #350: Models are broken when inserting from Fuel

Reka-Berci-Hajnovics, 2020-09-14:
I've tried to drag and drop several models from Fuel, but they're not loaded correctly.
The red hatchback model cannot find the wheel textures, the PR2_SDF17 has some problems at the joint description, and the x4_uav model doesn't give any errors, but it is exploded.
[image]
[image]
[image]

I've copied the empty.sdf file in my own folder and changed the rendering to ogre, as described here: [https://github.com/ignitionrobotics/ign-gazebo/issues/344#issuecomment-691182579](https://github.com/ignitionrobotics/ign-gazebo/issues/344#issuecomment-691182579)
When I'm running the updated file the ignition collapses as soon as I try to insert any model from Fuel.
[image]

chapulina, 2020-10-10:
Some problems loading models are issues with the model, others are issues on Ignition. This is the flow to address these issues:

1. Is it a **bug in Ignition**, such as a crash or some mesh that's loading incorrectly?
    * **Ticket an issue** in the appropriate library if there isn't yet an issue for that specific bug.
1. Is it a problem with the **model description** itself (i.e. it doesn't follow the guidelines in https://github.com/ignitionrobotics/ign-gazebo/pull/400)?
    * Is the models in the [Open Robotics](https://app.ignitionrobotics.org/OpenRobotics) account?
        * **Ticket an issue** on https://github.com/ignitionrobotics/docs/ if there isn't one yet.
    * Otherwise, contact the model owner.

That said, about the specific models you're trying to load:

* https://app.ignitionrobotics.org/chapulina/fuel/models/hatchback_red: not an Open Robotics model (I'm the owner and I put experimental models in my account that may not work)
* https://app.ignitionrobotics.org/OpenRobotics/fuel/models/PR2_SDF17 and https://app.ignitionrobotics.org/OpenRobotics/fuel/models/X4%20UAV: it looks like you ran into a bug, but they load fine for me. Can you provide information about your system? What Ignition version, what operating system, graphics card...

claireyywang, 2020-10-12:
I was able to reproduce this on both Citadel and Dome (both source build) on Bionic 18.04. 

It looks like a sdformat bug. 

Here's the terminal output and screenshot: 

```
[Err] [SDFFeatures.cc:720] Asked to create a joint between links [torso_lift_link] as parent and [torso_lift_motor_screw_link] as child, but the child link already has a parent joint of type [RevoluteJoint].
[Err] [SDFFeatures.cc:720] Asked to create a joint between links [r_gripper_r_finger_tip_link] as parent and [r_gripper_motor_screw_link] as child, but the child link already has a parent joint of type [RevoluteJoint].
[Err] [SDFFeatures.cc:720] Asked to create a joint between links [r_gripper_l_finger_tip_link] as parent and [r_gripper_motor_screw_link] as child, but the child link already has a parent joint of type [RevoluteJoint].
[Err] [SDFFeatures.cc:720] Asked to create a joint between links [r_gripper_r_parallel_link] as parent and [r_wrist_roll_link] as child, but the child link already has a parent joint of type [RevoluteJoint].
[Err] [SDFFeatures.cc:720] Asked to create a joint between links [r_gripper_l_parallel_link] as parent and [r_wrist_roll_link] as child, but the child link already has a parent joint of type [RevoluteJoint].
[Err] [SDFFeatures.cc:720] Asked to create a joint between links [r_gripper_r_parallel_link] as parent and [r_gripper_r_finger_tip_link] as child, but the child link already has a parent joint of type [RevoluteJoint].
[Err] [SDFFeatures.cc:720] Asked to create a joint between links [r_gripper_l_parallel_link] as parent and [r_gripper_l_finger_tip_link] as child, but the child link already has a parent joint of type [RevoluteJoint].
[Err] [SDFFeatures.cc:720] Asked to create a joint between links [r_gripper_r_finger_tip_link] as parent and [r_gripper_l_finger_tip_link] as child, but the child link already has a parent joint of type [RevoluteJoint].
[Err] [SDFFeatures.cc:720] Asked to create a joint between links [l_gripper_r_finger_tip_link] as parent and [l_gripper_motor_screw_link] as child, but the child link already has a parent joint of type [RevoluteJoint].
[Err] [SDFFeatures.cc:720] Asked to create a joint between links [l_gripper_l_finger_tip_link] as parent and [l_gripper_motor_screw_link] as child, but the child link already has a parent joint of type [RevoluteJoint].
[Err] [SDFFeatures.cc:720] Asked to create a joint between links [l_gripper_r_parallel_link] as parent and [l_wrist_roll_link] as child, but the child link already has a parent joint of type [RevoluteJoint].
[Err] [SDFFeatures.cc:720] Asked to create a joint between links [l_gripper_l_parallel_link] as parent and [l_wrist_roll_link] as child, but the child link already has a parent joint of type [RevoluteJoint].
[Err] [SDFFeatures.cc:720] Asked to create a joint between links [l_gripper_r_parallel_link] as parent and [l_gripper_r_finger_tip_link] as child, but the child link already has a parent joint of type [RevoluteJoint].
[Err] [SDFFeatures.cc:720] Asked to create a joint between links [l_gripper_l_parallel_link] as parent and [l_gripper_l_finger_tip_link] as child, but the child link already has a parent joint of type [RevoluteJoint].
[Err] [SDFFeatures.cc:720] Asked to create a joint between links [l_gripper_r_finger_tip_link] as parent and [l_gripper_l_finger_tip_link] as child, but the child link already has a parent joint of type [RevoluteJoint].

```
[image]

chapulina, 2020-10-13:
Thanks for checking @claireyywang . I think there are 2 separate issues. I also get the error messages, but I don't see any broken behaviour on the robot, pressing play all the joints seem to be in place.

The most pressing issue are the distorted meshes, and I'm not sure how to reproduce that.

Reka-Berci-Hajnovics, 2020-10-13:
> Some problems loading models are issues with the model, others are issues on Ignition. This is the flow to address these issues:
> 
> 1. Is it a **bug in Ignition**, such as a crash or some mesh that's loading incorrectly?
>    
>    * **Ticket an issue** in the appropriate library if there isn't yet an issue for that specific bug.
> 2. Is it a problem with the **model description** itself (i.e. it doesn't follow the guidelines in #400)?
>    
>    * Is the models in the [Open Robotics](https://app.ignitionrobotics.org/OpenRobotics) account?
>      
>      * **Ticket an issue** on https://github.com/ignitionrobotics/docs/ if there isn't one yet.
>    * Otherwise, contact the model owner.
> 
> That said, about the specific models you're trying to load:
> 
> * https://app.ignitionrobotics.org/chapulina/fuel/models/hatchback_red: not an Open Robotics model (I'm the owner and I put experimental models in my account that may not work)
> * https://app.ignitionrobotics.org/OpenRobotics/fuel/models/PR2_SDF17 and https://app.ignitionrobotics.org/OpenRobotics/fuel/models/X4%20UAV: it looks like you ran into a bug, but they load fine for me. Can you provide information about your system? What Ignition version, what operating system, graphics card...

I'm using Ignition Citadel, Ubuntu 20.04, and my graphics card is: NVIDIA Corporation GP106BM [GeForce GTX 1060 Mobile 6GB]

azeey, 2020-10-13:
The distorted meshes appear to be similar to https://github.com/osrf/subt/issues/438. Can you try testing with the instructions in [this](https://github.com/osrf/subt/issues/438#issuecomment-640290247) or [this](https://github.com/osrf/subt/issues/438#issuecomment-665161905) comments?

The `[Err] [SDFFeatures.cc:720] Asked to create a joint between links [torso_lift_link] as parent and [torso_lift_motor_screw_link] as child, but the child link already has a parent joint of type [RevoluteJoint].` errors are most likely due to the fact that ign-gazebo doesn't currently support kinematic loops with DART.

chapulina, 2020-10-13:
Oh so this is another locale issue. Here are other related issues: https://github.com/ignitionrobotics/ign-rendering/issues/101, https://github.com/ignitionrobotics/ign-rendering/issues/136

claireyywang, 2020-10-13:
@anindex fyi

Reka-Berci-Hajnovics, 2020-10-14:
> The distorted meshes appear to be similar to [osrf/subt#438](https://github.com/osrf/subt/issues/438). Can you try testing with the instructions in [this](https://github.com/osrf/subt/issues/438#issuecomment-640290247) or [this](https://github.com/osrf/subt/issues/438#issuecomment-665161905) comments?
> 
> The `[Err] [SDFFeatures.cc:720] Asked to create a joint between links [torso_lift_link] as parent and [torso_lift_motor_screw_link] as child, but the child link already has a parent joint of type [RevoluteJoint].` errors are most likely due to the fact that ign-gazebo doesn't currently support kinematic loops with DART.

I've tried to run Ignition with an english locale like suggested in the links above and it worked! Thanks!
I haven't changed the whole system locale I just used the following command to run Ignition:
```
LC_ALL=C LANG=C ign gazebo empty.sdf
```
I still receive some errors, but the meshes are rendered properly.
[image]

chapulina, 2020-10-16:
Closing this issue. Summary:

* The hatchback model in the `chapulina` account is experimental
* The PR2 and X4 distortions are locale issues that are being tracked on https://github.com/ignitionrobotics/ign-rendering/issues/136

closed 2020-10-16 by chapulina