gazebosim/gz-sim #353: Citadel crashes when actor is added to the world

Reka-Berci-Hajnovics, 2020-09-16:
I was following the [Actors tutorial](https://ignitionrobotics.org/docs/citadel/actors). When  I try to run `ign gazebo actor_demo.sdf` ignition crashes and I get the following an error message.

I've tried to edit the SDF file to use `ogre` instead of `ogre2`, but in both cases, I get the same error:
[image]

I'm on Ubuntu 20.04.

chapulina, 2020-09-17:
I can't reproduce this failure on 18.04 or 20.04.

Can you see if upgrading all libraries fixes the issue?

```
sudo apt update
sudo apt upgrade
```

Also try removing the actor from your cache:

`rm -rf ~/.ignition/fuel/fuel.ignitionrobotics.org/Mingfei/models/actor/`

If none of these work, can you list all the version of Ignition libraries that you're running?

`apt list | grep ignition`

Reka-Berci-Hajnovics, 2020-09-17:
I've tried all the steps, but it didn't work.

The list of Ignition libraries:
`ignition-citadel/unknown,focal,focal,now 1.0.0-1~focal amd64 [installed]
ignition-tools/unknown,focal,focal,now 1.0.0-1~focal amd64 [installed,automatic]
libignition-cmake-dev/focal 0.6.1-1ubuntu1 amd64
libignition-cmake2-dev/unknown,now 2.5.0-1~focal amd64 [installed,automatic]
libignition-common-dev/focal 1.1.1-1build1 amd64
libignition-common3-av-dev/unknown,now 3.6.1-1~focal amd64 [installed,automatic]
libignition-common3-av/unknown,now 3.6.1-1~focal amd64 [installed,automatic]
libignition-common3-core-dev/unknown,now 3.6.1-1~focal amd64 [installed,automatic]
libignition-common3-dev/unknown,now 3.6.1-1~focal amd64 [installed,automatic]
libignition-common3-events-dev/unknown,now 3.6.1-1~focal amd64 [installed,automatic]
libignition-common3-events/unknown,now 3.6.1-1~focal amd64 [installed,automatic]
libignition-common3-graphics-dev/unknown,now 3.6.1-1~focal amd64 [installed,automatic]
libignition-common3-graphics/unknown,now 3.6.1-1~focal amd64 [installed,automatic]
libignition-common3-profiler-dev/unknown,now 3.6.1-1~focal amd64 [installed,automatic]
libignition-common3-profiler/unknown,now 3.6.1-1~focal amd64 [installed,automatic]
libignition-common3/unknown,now 3.6.1-1~focal amd64 [installed,automatic]
libignition-common/focal 1.1.1-1build1 amd64
libignition-fuel-tools1-1/focal 1.0.0+dfsg4-5build2 amd64
libignition-fuel-tools1-dev/focal 1.0.0+dfsg4-5build2 amd64
libignition-fuel-tools4-dev/unknown,focal,focal,now 4.2.1-2~focal amd64 [installed,automatic]
libignition-fuel-tools4/unknown,focal,focal,now 4.2.1-2~focal amd64 [installed,automatic]
libignition-gazebo3-dbg/unknown 3.3.0-1~focal amd64
libignition-gazebo3-dev/unknown,focal,focal,now 3.3.0-1~focal amd64 [installed,automatic]
libignition-gazebo3-plugins/unknown,focal,focal,now 3.3.0-1~focal amd64 [installed,automatic]
libignition-gazebo3/unknown,focal,focal,now 3.3.0-1~focal amd64 [installed,automatic]
libignition-gui3-dev/unknown,focal,focal,now 3.3.0-1~focal amd64 [installed,automatic]
libignition-gui3/unknown,focal,focal,now 3.3.0-1~focal amd64 [installed,automatic]
libignition-launch2-dev/unknown,focal,focal,now 2.1.0-1~focal amd64 [installed,automatic]
libignition-launch2/unknown,focal,focal,now 2.1.0-1~focal amd64 [installed,automatic]
libignition-math4-dev/focal 4.0.0+dfsg1-5ubuntu1 amd64
libignition-math4/focal 4.0.0+dfsg1-5ubuntu1 amd64
libignition-math6-dbg/unknown 6.5.0-2~focal amd64
libignition-math6-dev/unknown,now 6.5.0-2~focal amd64 [installed,automatic]
libignition-math6-eigen3-dev/unknown,now 6.5.0-2~focal amd64 [installed,automatic]
libignition-math6/unknown,now 6.5.0-2~focal amd64 [installed,automatic]
libignition-msgs-dev/focal 1.0.0+dfsg1-5build2 amd64
libignition-msgs5-dev/unknown,focal,focal,now 5.3.0-1~focal amd64 [installed,automatic]
libignition-msgs5/unknown,focal,focal,now 5.3.0-1~focal amd64 [installed,automatic]
libignition-msgs/focal 1.0.0+dfsg1-5build2 amd64
libignition-physics2-core-dev/unknown,focal,focal,now 2.2.0-1~focal amd64 [installed,automatic]
libignition-physics2-dartsim-dev/unknown,focal,focal,now 2.2.0-1~focal amd64 [installed,automatic]
libignition-physics2-dartsim/unknown,focal,focal,now 2.2.0-1~focal amd64 [installed,automatic]
libignition-physics2-dev/unknown,focal,focal,now 2.2.0-1~focal amd64 [installed,automatic]
libignition-physics2-mesh-dev/unknown,focal,focal,now 2.2.0-1~focal amd64 [installed,automatic]
libignition-physics2-sdf-dev/unknown,focal,focal,now 2.2.0-1~focal amd64 [installed,automatic]
libignition-physics2-tpe-dev/unknown,focal,now 2.2.0-1~focal amd64 [installed,automatic]
libignition-physics2-tpe/unknown,focal,now 2.2.0-1~focal amd64 [installed,automatic]
libignition-physics2-tpelib-dev/unknown,focal,now 2.2.0-1~focal amd64 [installed,automatic]
libignition-physics2-tpelib/unknown,focal,now 2.2.0-1~focal amd64 [installed,automatic]
libignition-physics2/unknown,focal,focal,now 2.2.0-1~focal amd64 [installed,automatic]
libignition-plugin-dbg/unknown 1.1.0-1~focal amd64
libignition-plugin-dev/unknown,focal,focal,now 1.1.0-1~focal amd64 [installed,automatic]
libignition-plugin/unknown,focal,focal,now 1.1.0-1~focal amd64 [installed,automatic]
libignition-rendering3-core-dev/unknown,focal,focal,now 3.1.0-1~focal amd64 [installed,automatic]
libignition-rendering3-dev/unknown,focal,focal,now 3.1.0-1~focal amd64 [installed,automatic]
libignition-rendering3-ogre1-dev/unknown,focal,focal,now 3.1.0-1~focal amd64 [installed,automatic]
libignition-rendering3-ogre1/unknown,focal,focal,now 3.1.0-1~focal amd64 [installed,automatic]
libignition-rendering3-ogre2-dev/unknown,focal,focal,now 3.1.0-1~focal amd64 [installed,automatic]
libignition-rendering3-ogre2/unknown,focal,focal,now 3.1.0-1~focal amd64 [installed,automatic]
libignition-rendering3/unknown,focal,focal,now 3.1.0-1~focal amd64 [installed,automatic]
libignition-sensors3-air-pressure-dev/unknown,focal,focal,now 3.0.0-1~focal amd64 [installed,automatic]
libignition-sensors3-air-pressure/unknown,focal,focal,now 3.0.0-1~focal amd64 [installed,automatic]
libignition-sensors3-altimeter-dev/unknown,focal,focal,now 3.0.0-1~focal amd64 [installed,automatic]
libignition-sensors3-altimeter/unknown,focal,focal,now 3.0.0-1~focal amd64 [installed,automatic]
libignition-sensors3-camera-dev/unknown,focal,focal,now 3.0.0-1~focal amd64 [installed,automatic]
libignition-sensors3-camera/unknown,focal,focal,now 3.0.0-1~focal amd64 [installed,automatic]
[... cut ...]
libignition-transport8-dbg/unknown,focal,focal 8.1.0-1~focal amd64
libignition-transport8-dev/unknown,focal,focal,now 8.1.0-1~focal amd64 [installed,automatic]
libignition-transport8-log-dev/unknown,focal,focal,now 8.1.0-1~focal amd64 [installed,automatic]
libignition-transport8-log/unknown,focal,focal,now 8.1.0-1~focal amd64 [installed,automatic]
libignition-transport8/unknown,focal,focal,now 8.1.0-1~focal amd64 [installed,automatic]
ros-foxy-dolly-ignition/focal,now 0.3.0-1focal.20200911.235014 amd64 [installed,automatic]
`

chapulina, 2020-09-17:
Thank you for the info. It looks like you have everything up to date. Not sure what could be the problem. Maybe @iche033 has some ideas.

iche033, 2020-09-17:
I can't reproduce the issue either. The error looks like it's passing nan values to ogre. Maybe it's a locale issue? Could you try doing `export LC_NUMERIC="C"` first before running ign gazebo?

Reka-Berci-Hajnovics, 2020-10-05:
It works! Thanks :) Running `export LC_NUMERIC="C"` before running ignition gazebo solved it.

closed 2020-10-05 by Reka-Berci-Hajnovics

chapulina, 2020-10-05:
Glad to hear it worked, @Reka-Berci-Hajnovics . Please consider adding a note about that to the troubleshooting section: https://ignitionrobotics.org/docs/citadel/install_ubuntu_src#troubleshooting