gazebosim/gz-sim #360: How to set sensor depth topics independently ?

doisyg, 2020-09-20:


doisyg, 2020-09-20:
Sensor depth outputs two topics: an depth image and a point cloud, however there is only one `<topic>` attribute under `<sensor>`. 
Right now the depth image is outputted on  `/topic` and the pointcloud on `/topic/points` 
I looked quickly at the code, independently setting the 2 topics is not implemented. I could PR however, I am not sure what would be the best approach. Adding a <topic_points> attribute would imply modifying also the sdf format, right ? Or maybe implementing a remap feature ? But at which level ?

doisyg, 2020-09-20:
Ideally providing the equivalent of the gazebo libgazebo_ros_openni_kinect.so plugin attributes:
```
<depthImageTopicName>
<depthImageCameraInfoTopicName>
<pointCloudTopicName>
```

chapulina, 2020-09-20:
> Adding a <topic_points> attribute would imply modifying also the sdf format, right ?

Correct

> Or maybe implementing a remap feature ? But at which level ?

If you want it to be very generic, I think it would go at the `ign-transport` level. 

---

I don't have a good idea of how to solve this without touching SDF. This feels a bit hacky, but we could support multiple topics within the `<topic>` tag and put a separator between them?